nunobrum/spicelib #2: RawReader .AC simulation results

nunobrum, 2023-09-17:
RawReader doesn't read .AC simulation files.

hsiljedal, 2023-09-17:
Ah, thanks for showing me the preferred way for reporting bugs or other issues.

From my e-mail (sorry for that):

"
A .qraw file typically includes this in its header
Command: QSPICE64, Build Sep 13 2023 09:01:15

and I notice your code only looking for QSPICE. That might cause trouble.
I tried to change my .qraw file header to QSPICE, but this did not help.

Both 'frequency' axis and node voltages are read wrong.
"

nunobrum, 2023-09-17:
The clause "QSPICE" in command assures that any variation of *QSPICE* can be accepted.
I think the problem is on the parsing of the binary data. This was already the case for the .TRAN. 
Unfortunately, QSPICE doesn't follow the same RAW format as all other spice simulators.

hsiljedal, 2023-09-18:
It does not work when I use 'traces_to_read' to RawRead()

from line 439:
            elif (traces_to_read == "*") or (name in traces_to_read):
                if has_axis:  # Reads data
                    trace = TraceRead(name, var_type, self.nPoints, self.axis, numerical_type)
                else:
                    # If an Operation Point or Transfer Function, only one point per step
                    trace = TraceRead(name, var_type, self.nPoints, None, 'real')
            else:
                trace = DummyTrace(name, var_type)

Traceback (most recent call last):
  File "c:\Programs\Python\env\km\Lib\site-packages\pyunav\ltc\ltc_raw2csv.py", line 67, in <module>
    LTR = RawRead("C:\\Users\\havardmy\\Projects\\hs\\qspice\\doc\\tb_atab40v_tx_LIN_AC_FB_10k_100k.qraw","V(out)")
  File "C:\Programs\Python\env\km\lib\site-packages\spicelib\raw\raw_read.py", line 473, in __init__
    if trace.numerical_type == 'double':
AttributeError: 'DummyTrace' object has no attribute 'numerical_type'

But when I read all traces (i.e. traces_to_read=='*') **everything works fine**.
Thank you for looking into this so quickly!

nunobrum, 2023-10-11:
I think this can be closed now. The new spicelib version can digest all QSpice simulation types.

closed 2023-10-11 by nunobrum