nunobrum/spicelib #4: Some suggestions.

xgzi, 2023-09-19:
Hi.
I find there's another repository in GitHub that is similar to 'spicelib'.
I found its ideas interesting.
[PySpice](https://github.com/PySpice-org/PySpice)
[A PPT illustrating its idea](https://www.slideshare.net/PoleSystematicParisRegion/pyparis2017-circuit-simulation-using-python-by-fabrice-salvaire)

**The basic idea:** in general, doing a spice simulation has three steps: 1. edit circuit; 2. run simulation; 3. data analysis.
The PySpice lets the user do the first and last steps in Python.
However, firstly, it can only support ngspice and xyce. The second reason I dislike it is that I have to learn its internally defined method to edit circuits rather than use general spice language. And the code complexity and bugs.

---

It's possible to learn from it and improve spicelib. Here are my suggestions:
(They may be wrong.)

1. Currently, 'spicelib' relies too much on LTspice in circuit editing. For example, to start a simulation, I have to draw a circuit in LTspice to form an 'asc' file and then convert it in LTspice to form a 'net' file, and finally, the 'SpiceEditor' class can read this circuit.
> but in ngspice and xyce, as I know, the user can directly create their circuit by writing codes. That is useful when simulating large circuits or doing version control.
I suggest providing a way for the user to create a blank circuit netlist and directly manipulate it. #3 

2. Since you can read the asc file, is it possible to convert the asc into the net in Python? Each time I modified my asc circuit, I had to convert it to net manually.

3. The spicelib can provide data results in numpy array. It can convenience the data analysis and visualization for the user.
More examples and documentation can be built to show how to use numpy and matplotlib to do something more. (I may help)

nunobrum, 2023-09-19:
Hello @xgzi ,

thanks for your suggestion and pull-request. I've already seen PySpice before and was interested by that approach.
However, I have mixed feelings in regards to describing a whole circuit in Python. It was simply not designed for that purpose,
and ends up being more akward than Spice itself.

Having said that, I do think that allowing to start a circuit from Python, and make macros using Spice primitives does make sense. Specially when instantiating hundreds of circuits with varying parameters.
I'll see if I can redirect your pull-request into the dev branch and will accept it. I'll probably add a bit more documentation to it and make some updates.
Don't worry, I'll credit you for the idea and initial implementation.

Thanks for your contribution,
Nuno

nunobrum, 2023-10-01:
Closing this issue as it was now merged into main branch.

closed 2023-10-01 by nunobrum