nunobrum/spicelib #8: Cant import LTspice from spicelib

AnasMalas, 2023-10-10:
After trying the same thing with your PyLTSpice library which you seem to be moving away from, I cannot get a file to run. 

I will preface this with that I am a beginner when it comes to python, so while im doing my best to read and understand, I could be missing something obvious. 

I am primarily trying to use https://spicelib.readthedocs.io/en/latest/modules/run_simulations.html, but 

`from spicelib import SimRunner, SpiceEditor, LTspice ` 

does not work, seems like this is because you removed LTspice from __init__ [here](https://github.com/nunobrum/PyLTSpice/commit/bf01f0ceb867f7aa840ca192b4d9f089b864360b)

chatGPT is trying to import [ltspice_pytool](https://github.com/DongHoonPark/ltspice_pytool) when I ask it, but my understanding is that your library isnt built on that one. Am I mistaken there?

Thank you for your time!

rliangcn, 2023-10-10:
This can solve your problem.
```
from spicelib.simulators.ltspice_simulator import LTspice
```

nunobrum, 2023-10-10:
Thanks  @rliangcn 
I'll add the missing statements back to the __init__.py

@AnasMalas sorry for the inconvinience.

closed 2023-10-10 by nunobrum

AnasMalas, 2023-10-12:
Thanks @rliangcn and @nunobrum. I appreciate that you updates the documentation :)