nunobrum/spicelib #9: Program stuck when running multiple simulations simultaneously.

rliangcn, 2023-10-10:
Hi,

I have a circuit which is a little complicated. When I run ltspice to simulate it in multiple tasks, I found it may stuck like this:
[image]

See the project in the attachment. To run this, please replace the ltspice exe path first.

**The key parameter** is `parallel_sim=4` in `SimRunner` instantiation. The larger value of it, the more possibility to reproduce the bug.
(I tried [1, 4, 16]. For '1', it can succeed to end; For '16', it stucked immediately.)

I guess that might be caused by the waiting and lock mechanism in multithread running. 

---
system:  Windows 10 22H2
Ltspice version: 17.0.37.0 (Jul 20 2023)

Thanks.
[example.zip](https://github.com/nunobrum/spicelib/files/12856353/example.zip)

nunobrum, 2023-10-11:
Hello !

I've tested this circuit with both parallel_sim=4 and parallel_sim=16 and it didn't stuck.
What I've noticed is that I wasn't able to write this while the simulations were running.
I've used the latest version of LTspice. I see that although you say you have a new LTspice version installed, 
the path only your script still points into an XVII version. The new versions have a LTSpice.exe executable.
Maybe you could try setting the path to the new LTspice.

Finally...
The KeyboardInterrupt message is the signature of the user sending a Ctrl+C event to the process console.
Does this ring a bell to you ?

Best Regards,
Nuno

rliangcn, 2023-10-12:
Hi Nuno, thank you very much!

Replacing XVII with LTspice 17 solved my problem!
I mistakenly thought that XVII 17 was the same as LTspice 17. The stuck problem seems to be from the XVII executable.

About the KeyboardInterrupt:
I have to Ctrl+C because the program is stuck; otherwise, I must wait about 600s for the program time out.

I also noticed that transitioning from one simulation run to another will interrupt the user typing text.

closed 2023-10-12 by nunobrum