nunobrum/spicelib #20: Linux Running

domovinavi, 2024-01-08:
Hello,

i wanted to ask if this was done purposly. 

When i start the Sallenkey.py Testcase for the WCA, the simulation starts and the terminal outputs Starting Simulation 1.
An empty LTSpice window opens and afterwards nothing happens. 
When i close the LTSPice window, the output tells me the log files were not found etc.
But is this behaviour done on purpose? Do i need to manually click the run button on the .asc and so on?
This would defeat the purpose.

Im running spicelib 0.8 on ubuntu 22.04

Thanks and best regards,
Dominik

nunobrum, 2024-02-02:
Hello Dominik,

I was hoping to have an opportunity sooner to try spicelib in Linux, but so far I wasn't able to. 
Fact is, I don't have a readily linux setup available to test this.
The Linux implementation was done by other GitHub users that kindly pushed back their edits back to the upstream server.
I'm sorry that LTspice is not working properly in your linux, but, I can't really assist you now and I don't want to delay this answer much longer.

Thanks for understanding,
Nuno

domovinavi, 2024-02-02:
Hello Nuno,

thanks for your answer!  I set up a VM with windows and there it works fine. Small workaround but I guess it is what it is.
Thanks for your efforts!
Domi

hakan-demirli, 2024-02-04:
This behaviour is not specific to spicelib.
**I have tested:**
**Ltspice:** `XVII, latest`
**Wine**: `stable, staging, wayland`
**Runner:** `python, bash, shell`
**PIP Packages:** `PyLTSpice, spicelib, ltspice`

In each and every one of them you have to manually open the asc file after ltspice window pops up. Since most simulation automation libraries parallelize simulations, you have to click hundreds of times before all simulations start.

I think the root cause is either wayland or wine. Since, there have been bunch of scripts and libraries created in the past that claims to be running on Linux.

Until someone finds a real workaround/fix, I will be using a spicelib server inside a Windows VM.

nunobrum, 2024-02-08:
At this point I'd better state upfront, that I don't give support to the problems related to the Linux environment. 
There there is simply too much that can go wrong.

Draxiss314, 2024-03-08:
I'm having the same issue, on version 5.2, using LTSpiceXVII, on Pop!_OS 22.04, and running through PyCharm with Python 3.10. 

I have finally resorted to just importing os and running `os.system('wine ~/.wine/drive_c/Program\ Files/LTC/LTspiceXVII/XVIIx64.exe dc_circuit.net')` if I want to simulate a netlist called 'dc_circuit.net' located in the directory of my script. 

I still have to manually close it, but it's functional. Effectively what's happening is the final argument isn't being called, as far as I can tell.

Draxiss314, 2024-03-11:
Workaround Addendum: If you want to run and then close ltspice, you should run something like this: `os.system('wine ~/.wine/drive_c/Program\ Files/LTC/LTspiceXVII/XVIIx64.exe dc_circuit.net -Run -b')`. It seems that for very simple netlists they will be automatically simulated, but you need -run for anything more complicated. Not sure on the specifics. Also, -b doesn't apparently need a batch file, so it works for ensuring you don't need to manually close the window each time. I'm not super good at arguments so if there's a more efficient argument that does what I need do let me know.

nunobrum, 2024-03-16:
Interesting. I can't imagine how that can possibly work. If checking LTspice documentation, the file to simulate should come after the -b option. See [https://ltwiki.org/LTspiceHelp/LTspiceHelp/Command_Line_Switches.htm](https://ltwiki.org/LTspiceHelp/LTspiceHelp/Command_Line_Switches.htm)
Also interesting to find in this list the -wine option. Can any of you with a Linux installation check what it does and report back if this somehow fixes the problems you have.

hb020, 2024-05-20:
Can confirm there are problems with running it from pyltspice. I'm using MacOS + wine, which is rather close to Linux + wine.

I switched to 
```python
        # the ltspice installation in wine is for my user and not for all users. If it is for all users, there would be "Program\ Files" in it.
        # You will need to fully expand the path to ltspice I found, leaving ~ in it can give problems.
        ltspice_path = os.path.expanduser(f"~/.wine/drive_c{os.path.expanduser('~')}/AppData/Local/Programs/ADI/LTspice/LTspice.exe")
        spice_exe = ['wine', ltspice_path, '-alt', '-Run', '-b']
        # run the command and capture the output to a log file. wine is verbose, and has many "normal errors". Look in that log file if your simulation does not run correctly.
        with open(processlogfile, "w") as outfile:
            subprocess.run(spice_exe + [netlistfile], stdout=outfile, stderr=subprocess.STDOUT)
```
(I need -alt) This results in the actual command being:
```bash
wine /users/myname/.wine/drive_c/Users/myname/AppData/Local/Programs/ADI/LTspice/LTspice.exe -alt -Run -b circuit.net
```

nunobrum, 2024-05-20:
Hi @hb020 ,

there are two interesting things on your code snippet. The first is that you redirect the stdout into a log file. 
Do you get in wine+ltspice.exe any useful information on that file that is worthwhile extracting?

Second thing is that you are not using the Mac OSX native LTspice application.  Are there any advantages in using wine+ltpice.exe instead of the LTspice.app ?

Cheers,
Nuno

hb020, 2024-05-20:
The stdout+stderr log will show problems with the command and the installation of ltspice. Nothing much about the simulation. But it can be handy at start. Much better to log it than showing the lot of ltspice and wine "normal" errors and warning messages that really mean nothing.

I do not use the LTspice.app, because I had problems with it, I remember it losing parts of the file name. Was planning to test that again (now I'm looking deeper into the asceditor)

hb020, 2024-06-01:
@marmeladapk : See my remarks in #51

closed 2024-06-16 by hb020