nunobrum/spicelib #21: Error: Worst Case Analysis and Sensitivity

domovinavi, 2024-01-08:
Hello again, 

i wanted to tell you that there is a bug inside the tolerance_deviations.py. You might check it out.

    self.log_data.stepset = {'run': [round(val.real) for val in self.log_data.dataset['run']]}
                                           ^^^^^^^^
AttributeError: 'str' object has no attribute 'real'

i tried by taking val as a float, but it seems that LTComplex classes cant be conversed to floats. Maybe you can check this out.

Best regards,
Dominik

nunobrum, 2024-01-12:
Hi @domovinavi 

can you give a bit more context on this error. Can you send me the log file that generates this error?
There is a couple of things I don't understand. First, the line you are referring to, is the line 251 of the tolerance_deviations.py file, right ?
Zooming out a bit, the previous line is :
```python
if isinstance(self.log_data.dataset['run'][0], LTComplex):
    self.log_data.stepset = {'run': [round(val.real) for val in self.log_data.dataset['run']]}
else:
    self.log_data.stepset = {'run': self.log_data.dataset['run']}
```
This means that the first element of the run, is an LTcomplex, therefore it contains the 'real' attribute.
In order for it to fail, it would need to be a log file, where the first element is identified as complex and then one of the other subsequent values as a string.
For me the problem is comming from elsewhere in the code.

Will wait for you to provide me more information on this issue.

Best Regards,
Nuno

nunobrum, 2024-02-02:
Please feel free to reopen this when you have more information.

closed 2024-02-02 by nunobrum

domovinavi, 2024-02-02:
Ill be back on this topic later, when the project I'm working on prioritizes WCA and Sensitivity again. 
Thanks for your Efforts and Best regards,
Domi