nunobrum/spicelib #24: Issues using set_parameter()

J-escandon, 2024-01-30:
Hi,

I am running into an issue when using AscEditor.set_parameter().
I have a base .asc with already predefined parameters. When trying to modify this using the AscEditor.set_parameter() I do get two different results depending on the parameter being changed.

I have the parameter: .param T0 = 100u. Using `asc.set_parameter('T0', 1/5000)` gets the following in the .asc: .param T0=100u 200u
I have the paremeter .param Ceq = 10p. `asc.set_parameter('Ceq', 10e-12)` simply creates a new paremeter .param Ceq = 10p in a new line while keeping the original one and dus duplicating it.

nunobrum, 2024-02-02:
Hi There Javier,

I've tried to replicate your finding, but, in my installation it does work well.
Can you please tell me what version are you using, or try with the latest version.

Best Regards,
Nuno

nunobrum, 2024-02-25:
Closing issue.

closed 2024-02-25 by nunobrum