nunobrum/spicelib #25: Hardcoded localhost

hakan-demirli, 2024-02-05:
For a Windows 10 VM running in QEMU/KVM,  "localhost" ip/url is not reachable from the host machine.

https://github.com/nunobrum/spicelib/blob/ce7bb03a307323dd1fee91506668306914624056/spicelib/client_server/sim_server.py#L42

However, I pass an empty string it works.
```python
('', port),
```

I suggest adding an extra argument `--url` which has `localhost` as default if it is not specified.

nunobrum, 2024-02-08:
I agree with your suggestion. 
I'll include this in the next release.

nunobrum, 2024-02-16:
If I pass a blank to the SimpleXMLRPCServer class I get a message from Windows complaining that I am trying to access non-authorized locations. Therefore, in order to support all situations, I'm adding to the SimServer __init__ a parameter called host, which is defaulting to "localhost".
I think this way you can updated it on your calling code, and the library remains untouched.

nunobrum, 2024-02-25:
host added to the input parameters in version 1.0.0
https://github.com/nunobrum/spicelib/blob/6d11c45779acd841309284ff985b0b5f75a576bf/spicelib/client_server/sim_server.py#L53

closed 2024-02-25 by nunobrum