nunobrum/spicelib #26: Send multiple files to server

hakan-demirli, 2024-02-05:
Currently sim_client only allows only one circuit file to be sent to the server. However, some simulations consist of multiple files which are included/used in the main netlist. For example, models of transmission lines might be in another `asc` file which is included into the main simulation file.

https://github.com/nunobrum/spicelib/blob/ce7bb03a307323dd1fee91506668306914624056/spicelib/client_server/sim_client.py#L121

nunobrum, 2024-02-15:
I'm advancing with the implementation of this feature, but, the proposed implementation however simple it has some caveats.
First of all, it is probably not a good idea to transfer everything inside the directory, as there might have simulation raw files that are useless and will slow the transfer of data.
Secondly, it is possible that the user is using a symbol or library path that are in other directories.
Third, we need to support different simulators which use different different extensions.

So, I was imagining the following. Instead of passing a simple circuit_dir, the user should supply an iterable providing all the dependencies. This iterable can be a manually list (or tuple) of all the required files. or using the os.listdir() can be used to construct this iterator.
For example:
`(dep for dep in os.listdir('tests') if dep.endswith('.lib'))`

Later, I can implement an iterator on the Editor class itself that gets all dependencies.

What do you say? Is this something that can work for you?

hakan-demirli, 2024-02-15:
Sure, I didn't think about library and symbol paths. This solves all of these problems and much more extendable.

nunobrum, 2024-02-25:
Integrated into 1.0.0

closed 2024-02-25 by nunobrum