nunobrum/spicelib #29: .log file now contains the version info in the first line as of LTspice 24.0.0

HvQ, 2024-02-09:
Hi Nuno,
there is a small change with the latest LTspice update (24.0.0+). The log file now contains the version info in the first line and not "Circuit:" anymore. Changing line 304 in _spicelib/log/ltsteps.py_ to "self.encoding = detect_encoding(log_filename, "LTspice")" and line 758 in _spicelib/raw/raw_read.py_ to "encoding = detect_encoding(logfile, "LTspice")" solved the issue. Maybe you could update this in the next release. Thank you for this amazing project.

**New:**

```
LTspice 24.0.9 for Windows
Circuit: * RDSon_template.net
Start Time: Fri Feb  9 10:11:48 2024
solver = Normal
Maximum thread count: 8
tnom = 27
temp = 27
method = modified trap

rdson2v5: v(drain)/i(v0)=5.24717 at 2.5
rdson4v5: v(drain)/i(v0)=2.59594 at 4.5
rdson10v: v(drain)/i(v0)=2.20168 at 10

Total elapsed time: 0.113 seconds.
```

**Old:**

```
Circuit: * RDSon_template.net

rdson2v5: v(drain)/i(v0)=5.24717 at 2.5
rdson4v5: v(drain)/i(v0)=2.59594 at 4.5
rdson10v: v(drain)/i(v0)=2.20168 at 10

Date: Thu Feb  8 16:49:39 2024
Total elapsed time: 0.034 seconds.

tnom = 27
temp = 27
method = trap
totiter = 281
traniter = 0
tranpoints = 0
accept = 0
rejected = 0
matrix size = 21
fillins = 7
solver = Normal
Avg thread counts: 7.1/0.0/8.0/7.1
Matrix Compiler1: 1.37 KB object code size  0.3/0.3/[0.2]
Matrix Compiler2: off  [0.2]/0.4/0.2
```

nunobrum, 2024-02-09:
Thanks for the heads up. I was not yet aware of this update.
I'll make a quick update to fix this issue.

nunobrum, 2024-02-10:
The fix is now available and will support all LTspice versions

closed 2024-02-10 by nunobrum