nunobrum/spicelib #38: Run simulations example code

rtclay, 2024-04-02:
The Run Simulations multiprocessing example does not work for me.

First, the line 
`run_netlist_file = "{}_{}_{}.net".format(netlist.netlist_file.name, opamp, supply_voltage)`
refers to the `netlist_file` attribute of netlist, but I think this attribute has been changed to `circuit_file`.  Changing the code to circuit_file makes this part work.

Second, when I call the line
`runner.run(netlist, run_filename=run_netlist_file, callback=processing_data)`
I get the error:

> RunTask # 1:Simulation Raw file or Log file were not found
> RunTask # 2:Simulation Raw file or Log file were not found
> RunTask # 3:Simulation Raw file or Log file were not found
> RunTask # 4:Simulation Raw file or Log file were not found

(Actually it produces errors with no space between # and the number, but I don't want to make links to pull requests.)
Running it without a `run_filename` parameter works.  

I have not yet figured out why it is not working when I run it with a `run_filename` parameter.  I think it should be creating raw and log files based on the `run_netlist_file` parameter, but it appears to error out when it fails to find raw and log files.  I expect the run function to create raw and log files, so the fact that it errors when it fails to find those raw and log files doesn't make sense to me.

rtclay, 2024-04-02:
I think part of the issue is that I have created the netlist as an AscEditor instead of a SpiceEditor.  When RunTask asks AscEditor for the netlist file, the AscEditor happily gives it a .net file, but that .net file doesn't exist.  Instead, the AscEditor has created a .asc file.

rtclay, 2024-04-02:
The example still doesn't work for me as written, because SpiceEditor is unable to open a file with the .asc suffix.

I resolved my problem by using AscEditor to open the .asc file, and then changing the `run_netlist_file` to use a .asc suffix rather than a .net suffix.

nunobrum, 2024-04-14:
I don't understand which of the examples are you referring to. I have just tested all examples on the spicelib and they all work on my installation.
Note that if you are using examples of the PyLTspice in the spicelib, it will not work.

nunobrum, 2024-05-09:
Closing the issue for lack of clarificaitons.

closed 2024-05-09 by nunobrum