nunobrum/spicelib #44: Vertical directives (TEXT) fail to load in AscEditor

hb020, 2024-05-19:
...because

```python
TEXT_REGEX = re.compile(r"TEXT (-?\d+)\s+(-?\d+)\s+(Left|Right|Top|Bottom)\s(\d+)\s*(?P<type>[!;])(?P<text>.*)",
                        re.IGNORECASE)
```
does not detect VLeft, VRight,...

Since the relevant code is undergoing movements (TEXT_REGEX is in asc_editor.py in the public version, but it got moved to ltspiceutils.py recently in the repo), and it looks like Work In Progress, I don't want to disturb that work. 

But be aware that you are missing text orientation cases.

nunobrum, 2024-05-19:
Nice catch. 
Yes. It's kind of a work in progress. I am fixing things as I find them. 
Anyway, I have decided to publish the code so people can try it out and perhaps make good use of it.

Thanks for your contribution,
Nuno

hb020, 2024-05-20:
Seen your commit. You are still missing Center, VCenter, VTop and VBottom if you want to be complete.

nunobrum, 2024-05-20:
After painstakingly seeing all available options in LTSpice, I figured that Invisible was also missing.

hb020, 2024-05-20:
yep, indeed. Good catch.

hb020, 2024-05-30:
For the record, this is corrected. Found some edge cases and corrected them. See branch "asceditor-encoding-fix-https://github.com/nunobrum/spicelib/issues/45" on top of branch v1.1.3.
Will close this when the branch is merged back into main.

closed 2024-06-03 by hb020