nunobrum/spicelib #45: AscEditor refuses to load MacOS asc files

hb020, 2024-05-19:
... as the file encoding is UTF-16 LE, without BOM. As a result, you get a NotImplementedError exception in reset_netlist.

There is no easy way out, one needs to test if the file is "normal", and it not, open with utf16-LE encoding.

Another way is to test and rewrite the file. For that, I use the code below:

```python
def check_and_correct_encoding(asc_file):
    
    # test if file exists.
    asc_file_path = Path(asc_file)
    if not asc_file_path.exists():
        raise FileNotFoundError(f"File {asc_file} not found")
    
    # test if format detection works.
    try:
        with open(asc_file_path, 'r') as asc_file:
            line = asc_file.readline().lower()
            if line.startswith("version"):
                return
    except:
        pass
    
    # NOT. so test in utf16-LE
    asc_file = open(asc_file_path, 'r', encoding='utf-16-le')
    line = asc_file.readline().lower()
    if line.startswith("version"):
        # Yes, that works. So rewrite the file
        # rewind to the start, read everything and close
        asc_file.seek(0)
        content = asc_file.read()
        asc_file.close()
        # now write the same content back to the file (overwriting the content)
        with open(asc_file_path, 'w') as asc_file:
            asc_file.write(content)
        return
    else:
        asc_file.close()
        raise NotImplementedError(f"Format not supported for ASC file {asc_file}")
```

It might be interesting to include that in spicelib or do automatic correction. 
Appended  a test file.
[encodingtest.asc.zip](https://github.com/nunobrum/spicelib/files/15369776/encodingtest.asc.zip)

nunobrum, 2024-05-19:
Hi @hb020 

If I understand the issue correctly, when a file has a character not recognized by the AscEditor, the script fails.
My first question is whether the file encoding is not always the same in the Mac. If this is the case, best would be to just check what is the platform the user has and force the encoding accordingly.

For the implementation, I suggest using the detect_encoding function that is included on the utils directory.

I'll check the file you've attached. I wonder if it would be an issue on my platform. (I work on Windows).

hb020, 2024-05-20:
LTSpice on Mac reads both UTF8 and UTF16-LE, but writes in UTF16-LE. And LTSpice on Windows does the reverse. 
The thing is, I am on Mac, so use the Mac version, but at the same time also use the Win version via Wine, because the Mac version does not support all command line options. So my PyLTSpice scripts on Mac have to manipulate files from both encodings. The detection based on the platform will not work for me. As a result, I decided to make my scripts compatible with both encodings.

nunobrum, 2024-05-26:
Ooops!  Somehow I missed your last answer.
Yes of course that I welcome any help I can have. 
The only rules I have are the normal ones:
 -  to try to have the same coding style across all sources. 
 - commits are done on branches that are then merged back to main.

You can make pull-requests from your own repo, or I can add you as a collaborator to mine. As you wish.
Maybe the latter has some advantages. Just let me know.

Best Regards,
Nuno

hb020, 2024-05-27:
Hi, the latter has advantages indeed. I can work on:
* the MacOS and Linux compatibility
  * file encoding, using detect_encoding, like spiceeditor)
  * some of the smaller asceditor problems I saw (equivalence between exceptions)  
  * I'll take a look why I couldn't use your runner, and, if you want, add the process log as mentioned in https://github.com/nunobrum/spicelib/issues/20#issuecomment-2119699803

I'll leave you #46, as you are already working on it. but I will test it of course. 

About #47 or do you want me to take that as well? 

Tell me what branch should be targeted.  1.1.3 or another.

nunobrum, 2024-05-28:
I have added you to the repository. You are the first official collaborator, so, it might require some additional configuration to do. 
It would be great if you can dive into the Linux side, because, I haven't now a possibility to do it. 
You can fix the #47 . That is an easy one, however, please not that I have refactored all the way the regular expressions are being "assembled". 
One day I'll muster the courage to update this SPICE parser using REGEX to use a proper code parser (PyLex or somehting like that).
Yes, the target is 1.1.3. Please branch from there.

hb020, 2024-05-28:
ok, will do. Thanks.

hb020, 2024-05-28:
Related to #48 , if you want, I can pick this up. Same place. Just need to set up my dev environment.

nunobrum, 2024-05-28:
Cool. I am quite overloaded at this moment. Thanks.

hb020, 2024-05-28:
Just did first commit to a fix branch. Do not merge yet, still to be tested. Just to check if this branch setup and naming is OK with you.

hb020, 2024-05-28:
Corrected. See branch "asceditor-encoding-fix-#45" on top of branch v1.1.3.

closed 2024-06-02 by nunobrum