nunobrum/spicelib #47: Giga (G) and Tera (T) values are not supported

hb020, 2024-05-26:
When setting  ```set_component_value("R2","10k")``` on a resistor that has "1G" as value, the value becomes "10kG". 
On which LTSPice does not complain, but which leads to unexpected results.

Reason: 

the various regex strings in ```spice_editor.py``` have the following constructions:

* ```[0-9\.E+-]+(Meg|[kmuµnpf]``` (3 times)
* ```[0-9\.E+-]+[kmuµnpf]``` (1 time)

t and g are missing here, so that should become

* ```[0-9\.E+-]+(Meg|[kmuµnpftg]``` (3 times)
* ```[0-9\.E+-]+[kmuµnpftg]``` (1 time)

I tested that, and it works.

Also, ```base_editor.py``` is missing support for T and G.

I can create a PR for that if you want. I will then also test more, as there might be more issues with the suffixes.

nunobrum, 2024-05-26:
Thanks! This part is actually being refactored. I can update it.

Will close when the new 1.1.3 branch is integrated into the main.

hb020, 2024-05-30:
Corrected. See branch "asceditor-encoding-fix-https://github.com/nunobrum/spicelib/issues/45" on top of branch v1.1.3.
Will close this when the branch is merged back into main.

closed 2024-06-03 by hb020