nunobrum/spicelib #48: LTspice .asc import: Issue 'utf-8' codec can't decode byte 0xb5 in position 376: invalid start byte.

TorfromNor, 2024-05-28:
Hello,
Im not sure if others experience the same issue however when i try to load a .asc file with spicelib i get a error: 'utf-8' codec can't decode byte 0xb5 in position 376: invalid start byte.
- Tried on 2 different computers

I tried to changed the asc_editor.py: reset_netlist encoding to unicode_escape (not sure why this one works) now the spicelib import runs and works atleast on the surface. 
- However what i then found was that now in import values for u is changed to µ and inside ltspice treated as  Âµ which then changes the results. allot so a new bug likely due to the change in import. (
- If i use spicelib to change values after the import of netlist and then export the values become correct and simulation.

I attached a zip with a ltspcie file where i have the issue and a jupyter notebook of the issue and a few things i tried. (if that works)

I could not figure out the problem or why i am not able to import with utf-8. (the utf-8 is likely the root issue)

Any idea what the issue is? and if it is only me which experiences it with .asc files? (latest LTspice version as of 28.05.2024 incase it is related with the new 17 version of LTspice x64 24.0.12)

[Spicelib utf-8 issue.zip](https://github.com/nunobrum/spicelib/files/15469634/Spicelib.utf-8.issue.zip)

hb020, 2024-05-28:
The file is encoded in iso8859-1, not UTF-8

nunobrum, 2024-05-28:
Moving forward we should probably use the detect encoding function on the AscEditor just in case.

hb020, 2024-05-28:
Corrected. See branch "asceditor-encoding-fix-#45" on top of branch v1.1.3. 
Will close this when the branch is merged back into main.

TorfromNor, 2024-05-29:
Thank you! I tested "asceditor-encoding-fix-https://github.com/nunobrum/spicelib/issues/45" and this solved the issue.

Just a final note.
Based on your comments i also realized the issue i faced was purely when i ran a jupyter notebook inside vscode. 
- Python script running the simulation script in vscode worked, however running a jupyter notebook in vscode got encoding issues.
- Jupyter notebook running the simulation script outside Vscode had no encoding issues.

Anyway it works fine now for the above with the new branch.

closed 2024-06-03 by hb020