nunobrum/spicelib #61: Regex error for simple netlist

mikekidner, 2024-07-17:
I've tried 
`netlist = SpiceEditor("example.net", encoding='utf-8')`
on a simple netlist (example.net) that contains only this:
```
*
R1 N001 0 R
C1 N001 0 C
L1 N001 0 L
V1 N001 0 V
.BACKANNO
.END
```
I get this error:

`raise UnrecognizedSyntaxError(line, regex.pattern)
spicelib.editor.spice_editor.UnrecognizedSyntaxError: Line: "R1 N001 0 R
" doesn't match regular expression "^(?P<designator>R§?\w+)(?P<nodes>(\s+\S+){2})(?P<model>\s+\w+)?\s+(R=)?(?P<value>(?P<formula>{)?(?(formula).*}|[-+]?[0-9]*\.?[0-9]+([eE][-+]?[0-9]+)?(Meg|[kRmuµnpfgt])?\d*))(?P<params>(\s+\w+\s*(=\s*[\w\{\}\(\)\-\+\*\/%\.]+)?)*)?.*$"`

Am I missing something obvious, or is this an actual issue? (the net file is encoded as UTF-8). 
Am I running LT-SPice on Mac OSX natively

nunobrum, 2024-07-17:
The R is not a valid resistance value. 
Same applies to all the other components. You need to give the simulator a value for these passives and a value or a valid primitive for the voltage source.

closed 2024-07-17 by nunobrum

mikekidner, 2024-07-17:
Thanks, that makes sense. I was going to fill those in via python, but they need to be valid first. Thanks again.