nunobrum/spicelib #62: Setting component value of OpAmp fails with SpiceEditor (ltspice)

hb020, 2024-07-23:
**Problem description:**
(latest github version of spicelib)

Imagine symbol with the following content in the .asc file:
```
SYMBOL OpAmps/opamp2 4288 -864 R0
SYMATTR InstName U1
SYMATTR Value OPAx189
```
that leads to the following info in the .net file:
```
XU1 N001 N002 +V -V N002 OPAx189
```

When calling 
```
netlist = spicelib.SpiceEditor(netlistname)
netlist.set_component_value("XU1", "OPAx140")
```
that results in 
```
XU1 N001 OPAx140 +V -V N002 OPAx189
```
instead of
```
XU1 N001 N002 +V -V N002 OPAx140
```
... as expected.

AscEditor works without problems.

**Cause**:

spicelib/editor/spce_editor.py:94
```
    # This implementation replaces everything after the 2 first nets
    'X': r"^(?P<designator>X§?\w+)(?P<nodes>(\s+\S+){1,99}?)\s+(?P<value>\w+)"
         r"(\s+params:)?" + PARAM_RGX + ".*$",
```
Well, it behaves as stated, not as expected. It is rather messy to do correctly, as X components can have any number of pins/nets. 

**Question:**

Would it be OK to let the regex look for the LAST value instead of the first?

hb020, 2024-07-24:
PR (#66) created

nunobrum, 2024-07-24:
I think this could be solved by adding a ? on the last part before the end of line marker. That makes it less eager to consume characters.

```
    'X': r"^(?P<designator>X§?\w+)(?P<nodes>(\s+\S+){1,99}?)\s+(?P<value>\w+)"
         r"(\s+params:)?" + PARAM_RGX + ".*?$",

```
I guess this bug was introduced by the last update I did on this :-(

hb020, 2024-07-25:
OK, will take a look. Will add some test cases in the unit tests as well.

nunobrum, 2024-07-25:
Yes. That is badly needed. I was confident that any issues would be caught by unit tests. 
What a mistake.

hb020, 2024-07-25:
Just pushed a new version of the regex. It is now
'''
^
(?P<designator>X§?\w+)
(?P<nodes>(\s+\S+){1,99})
\s+(?P<value>([^\s=]+(?:\s+[^\s=]+)*))
(?P<params>(?:\s+\S+\s*=\s*[^=\s]+)*)
\s*$
'''
Nodes: take as much as you can (eager)
Value: the last word that is not a key-value pair (because it will give back any key-value pair)
params: all key-value pairs (and key-value pairs only)
optional space (not any other character, as that will mess up)

I added some crooked test cases, see "opamptest.net" and "test_spice_editor.py".

It works now, on the cases that I see, but a rethink of all the regexes may be needed.

nunobrum, 2024-07-25:
It has been the Achilles heel of this project the regex approach. 

I started once a language parser but was quickly discouraged by it, has the spice language is not that uniform. Especially taking into account the different implementation of it. 

I was tempted to rewrite this whole implementing the object approach. But since I'm struggling to find time for the project, decided to minimize disruptions.

hb020, 2024-07-25:
Sure. Thought about that as well, like setting up a parser or a DSL, but that would mean a LOT of work. 
It works now, and it is not really homogenous because the netlist expressions are simply not homogenous. Yes, improvements are possible, let's just take them one at a time.

nunobrum, 2024-07-25:
> Just pushed a new version of the regex. It is now
> 
> ```
> 
> ^
> 
> (?P<designator>X§?\w+)
> 
> (?P<nodes>(\s+\S+){1,99})
> 
> \s+(?P<value>([^\s=]+(?:\s+[^\s=]+)*))
> 
> (?P<params>(?:\s+\S+\s*=\s*[^=\s]+)*)
> 
> \s*$
> 
> ```
> 
> See https://regex101.com/r/bjBbWh/1
> 
> 
> 
> * nodes: take as much as you can (eager)
> 
> * value: the last word that is not a key-value pair (because it will give back any key-value pair)
> 
> * params: all key-value pairs (and key-value pairs only)
> 
> * optional space (not any other character, as that will mess up)
> 
> 
> 
> I added some crooked test cases, see "opamptest.net" and "test_spice_editor.py".
> 
> 
> 
> It works now, on the cases that I see, but a rethink of all the regexes may be needed.

I'll have a look into all these pull requests this weekend

nunobrum, 2024-07-28:
Merged #66

closed 2024-07-28 by nunobrum