nunobrum/spicelib #64: AsyReader fails on lines with styles

hb020, 2024-07-24:
when an asy has a line like 

```LINE Normal -16 89 -16 96 2```

spicelib\spicelib\editor\asy_reader.py:71

```
                elif line.startswith("LINE"):
                    tag, weight, x1, y1, x2, y2 = line.split()
```
```
Exception has occurred: ValueError       (note: full exception trace is shown but execution is paused at: _run_module_as_main)
too many values to unpack (expected 6)
  File "...spicelib/spicelib/editor/asy_reader.py", line 72, in __init__
    tag, weight, x1, y1, x2, y2 = line.split()
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "...spicelib/spicelib/editor/asc_editor.py", line 248, in _get_symbol
    answer = AsyReader(asy_path)
             ^^^^^^^^^^^^^^^^^^^
  File "...spicelib/spicelib/editor/asc_editor.py", line 159, in reset_netlist
....
```

asc_editor handles this correctly.

Will provide a PR for this.

hb020, 2024-07-24:
PR (#65) created

nunobrum, 2024-07-25:
Weird bug. I wonder why this was not caught before. I did test this before committing.

hb020, 2024-07-25:
A fair part of the parser in asceditor is also applicable to asyeditor. It was just not ported. So I did that, and added warnings in the code. (and adapted the qsch writer)

nunobrum, 2024-07-28:
Merged #65

closed 2024-07-28 by nunobrum