nunobrum/spicelib #67: [improvement]: spicelib fails on some(?) LTspice installations

patzf, 2024-07-26:
OK first of all this is not really a bug of spicelib or PyLTSpirce but I found some strange behaviour today.
I have 3 different PCs , all on Win10 64 bit, all of them run the very same python3 version (3.11.5) and on all of the PCs there is a virtual environment in which the latest version of PyLTspice and spicelib are installed via an pip requirements file (to really have the same setup everywhere). Also, on all the machines, the very same version of LTSpice is running, i.e. the currently latest 24.0.12.

I have a script that runs a very simple transistor simulation and then does some plotting of the results. On two of the machines everything runs file. On one machine, spicelib has trouble reading the raw file. this looks like this

```
Traceback (most recent call last):
  File "C:\Users\myuser\printBjtQ1_modify_for_bjt_abs00.py", line 5, in <module>
    LTR = RawRead("bjtQ1_1.raw")

  File "C:\py311LTSpice\Lib\site-packages\spicelib\raw\raw_read.py", line 601, in __init__
    self._load_step_information(raw_filename)

  File "C:\py311LTSpice\Lib\site-packages\spicelib\raw\raw_read.py", line 758, in _load_step_information
    encoding = detect_encoding(logfile, "^(.*\n)?Circuit:")

  File "C:\py311LTSpice\Lib\site-packages\spicelib\utils\detect_encoding.py", line 77, in detect_encoding
    raise EncodingDetectError(f"Expected pattern \"{expected_pattern}\" not found in file:{file_path}")
spicelib.utils.detect_encoding.EncodingDetectError: Expected pattern "^(.*
)?Circuit:" not found in file:bjtQ1_1.log
```

Now, after having a look, I found that the very same LTSpice Version run by the the very same python script produces a different .log file on the machine that raises the error

The log file that WORKS looks like

```
LTspice 24.0.12 for Windows
Circuit: * C:\filepath\bjtQ1.asc
Start Time: Fri Jul 26 15:37:19 2024
Warning: Multiple definitions of model "2scr375p" Type: BJT
Warning: Multiple definitions of model "bc857b" Type: BJT
Warning: Multiple definitions of model "bc847c" Type: BJT
Warning: Multiple definitions of model "bc847b" Type: BJT
solver = Normal
Maximum thread count: 8
tnom = 27
temp = 27
method = modified trap
.step ib=8e-06
.step ib=5e-05
.step ib=0.0001
.step ib=0.0002
Total elapsed time: 0.012 seconds.
```

The log file that RAISES THE ERROR in SPICELIB looks like

```
LTspice 24.0.12 for Windows
Start Time: Fri Jul 26 15:37:37 2024
Warning: Multiple definitions of model "2scr375p" Type: BJT
Warning: Multiple definitions of model "bc857b" Type: BJT
Warning: Multiple definitions of model "bc847c" Type: BJT
Warning: Multiple definitions of model "bc847b" Type: BJT

   --- Expanded Deck Component Count ---
I's 1
Q's 1
V's 1
tot: 3

    --- Expanded Netlist ---
* C:\filepath\bjtQ1.asc
q1 n001 n002 0 0 2n2222
ib 0 n002 10u
vce n001 0 1
.model 2n2222 npn(is=1e-14 vaf=100 bf=200 ikf=0.3 xtb=1.5 br=3 cjc=8e-12 cje=25e-12 tr=100e-9 tf=400e-12 itf=1 vtf=2 xtf=3 rb=10 rc=.3 re=.2 vceo=30 icrating=800m mfg=nxp)
.dc vce 0 2.5 10m ib list 8u 50u 100u 200u
.end

solver = Normal
Maximum thread count: 12
tnom = 27
temp = 27
method = modified trap
.step ib=8e-06
.step ib=5e-05
.step ib=0.0001
.step ib=0.0002
Total elapsed time: 0.009 seconds.
```

As one can see the "corrupted" .log file does not seem to mention the word "Ciruit" before the file path and it looks in general "different". I have no idea why the same version of LTSpice should behave differently on different PCs if the same script is called and the same circuit is simulated (I sync the files between machines in a shared directory so I can be sure the files are the same, I also diffed the .asc files, they are identical on all machines).

So really this would be a question for the LTSpice Forum but has anyone here seen something like this also?

hb020, 2024-07-26:
can you diff the ltspice binary directories and the ltspice model/lib/symbol directories? And are their settings identical?

patzf, 2024-07-26:
Ok, after some more diffing I think I got the cause... I am not sure if this is a new default or there is somewhere an "old" config file in a path I am not aware of but LTSpice is using but the problem seemed to be the following setting (I found the difference by looking through all the items in the GUI)

[image]

I had to remove the check at "Generate Expanded Listing" so the log files looks the same. Quite logical If you know that setting but .. cannot remeber to have chnged that manually - I will try to get a full list of paths that LTSpice is using nd see where this setting came from...

Thanks!

nunobrum, 2024-07-28:
Hi @patzf ,

thanks for reaching out and investigating this. This setting is new to me. 
The best is to find a way to support this setting, otherwise there will be more people complaining about it.
I'll keep this open till this is fixed.

Best Regards,
Nuno

hb020, 2024-08-21:
merged in main, closing

hb020, 2024-09-24:
Still happens. See #94

hb020, 2024-09-24:
re-closing since this is a duplicate now.

closed 2024-09-24 by hb020