nunobrum/spicelib #69: Library search for asc_editor and spice_editor can be improved by looking at the simulator path

hb020, 2024-08-08:
Right now, I see these search paths:

```
AscEditor._lib_file_find():
os.path.expanduser("~/AppData/Local/LTspice/lib/sub"),
os.path.expanduser("~/Documents/LtspiceXVII/lib/sub"),
os.path.expanduser("~/AppData/Local/Programs/ADI/LTspice/lib.zip"),

AscEditor._asy_file_find():
os.path.expanduser("~/AppData/Local/LTspice/lib/sym"),
os.path.expanduser("~/Documents/LtspiceXVII/lib/sym"),

SpiceCircuit.get_subcircuit():
os.path.join(os.path.expanduser('~'), "Documents/LTspiceXVII/lib/sub", lib)

asc_to_qsch.convert_asc_to_qsch():
os.path.expanduser("~/AppData/Local/LTspice/lib/sym"),
os.path.expanduser("~/Documents/LtspiceXVII/lib/sym"),
```

2 issues show up:
1) not all paths are used everywhere
2) this is mainly guess work, while we know the path of the spice executable. The latter could be especially helpful when we have installations with wine, where these specific search paths are guaranteed to point nowhere (requiring the user to use `SpiceCircuit.add_library_search_paths()` or `SpiceEditor.LibSearchPaths.append()`), or potentially worse: point to a potentially outdated MacOS version without complaining.

I propose to tweak these searches to base the search on where the simulator really is installed. Will provide a PR.

hb020, 2024-08-08:
Just to trace the paths I found:

```
ltspice_path: /home/me/.wine/drive_c/users/me/Local Settings/Application Data/Programs/ADI/LTspice/LTspice.exe
lib_path: /home/me/.wine/drive_c/users/me/Local\ Settings/Application\ Data/LTspice/lib/

ltspice_path: /Users/me/.wine/drive_c/users/me/AppData/Local/Programs/ADI/LTspice/LTspice.exe
lib_path: /Users/me/.wine/drive_c/users/me/AppData/Local/LTspice/lib/
```

nunobrum, 2024-08-08:
The paths are not the same everywhere and there is a reason for that. In one case we are looking for symbols, another we are looking for libraries, and another for other .asc files.

hb020, 2024-08-08:
Yes, /sub, /sym endings are needed of course. Didn't make that explicit. The main point is that the number of paths to search is not identical everywhere. Working on a POC to automate lib path searches.

hb020, 2024-08-09:
Proposition:

### The situation right now

#### Simulators

* **LTspice** can use 3 library locations on Windows AFAIK, and a large variety under wine. It also has a 'factory default' set that is typically not updated after initial installation (`lib.zip`). The base path for the libraries under wine is known to the LTSpice runner.
* **others**: no idea in how many locations they can store the libraries. Also no idea if they can run under wine, but if so, the base path for the libraries under wine is known to the runner, in exactly the same way that applies to the LTspice runner.

#### Editors

**AscEditor** (for LTspice):
* hardcoded paths for the searching of:
  * `.lib` (/sub): 2 out of the 3 potential paths, + 'factory default', none compatible with wine
  * `.asy` (/sym): 2 out of the 3 potential paths, none compatible with wine
* custom paths can be added to the above searches via `AscEditor.add_library_paths(AscEditor, paths)`. In addition to the above, the custom paths will also be used for searching of:
  * `.asc`

**QschEditor** (for Qspice):
* hardcoded path for the searching of:
  * `.lib`: 1 path (out of ??), not compatible with wine
  * `.qsch`: 1 path (out of ??), not compatible with wine
*  custom paths can be added to the above searches via `QschEditor.lib_paths.extend(paths)` 

**asc_to_qsch** (for LTspice to Qspice conversion):
* hardcoded paths for the searching of:
  * `.asy` (/sym): 2 out of the 3 potential paths, none compatible with wine
* custom paths: not possible.

**SpiceEditor** (for LTspice, but potentially others):
* hardcoded path:
  * `.lib` (/sub): 1 path out of the 3 potential paths, not compatible with wine
* custom paths can be added to the above search via `SpiceEditor.add_library_search_paths(paths)` and `SpiceEditor.LibSearchPaths.append(paths)` (same base functionality, although the first is a bit more versatile). 

### Conclusions

* not all possible Windows paths are hardcoded, and coverage is varying among the functions that use hardcoded paths
* the runners could in theory help "hardcoding" the correct paths, and cover all platforms
* users of wine must use custom paths now
* the way to set custom paths is not standard among the editors/functions, and some do not even allow it

### Propositions

* standardise the way the editors get hardcoded search paths. meaning: 
  * get all relevant known paths to the correct places
  * allow seeding the correct paths from the runners, at editor instantiation (and/or a new function, TBD). This should only be needed for wine. Windows and native MacOS should work correctly by default.
* standardise the way the editors can get custom search paths:
  * a function name and signature should be chosen, out of the existing functions
  * and mark the others as deprecated
* and adapt the documentation of course

closed 2024-08-16 by nunobrum