nunobrum/spicelib #88: qsch_editor issue with sub circuits and ports not connected to anything

TorfromNor, 2024-09-14:
Hi,
The net editor in qspice assumes all nets is connected to something. This causes issues in subcircuits where a port have been left not connected, so for example non used pins causes the following error.  

sch_comp.ports = [self._find_net_at_pin(position, orientation, pin) for pin in pins]
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\torbj\OneDrive\Work\100 Python\03 Useful python codes to interface with other softwares\ltspice_Qspice_Ngspice\spicelib-development\spicelib\editor\qsch_editor.py", line 512, in _find_net_at_pin
    raise QschReadingError(f"Failed to find the net for {pin} in component in position {comp_pos}")
spicelib.editor.qsch_editor.QschReadingError: Failed to find the net for pin (-100,1000) (0,-200) 1 14 0 0x0 -1 "some_measurement" in component in position (3200, 200)

It would also be nice to include in the error message with which subcircuit the failure happens in so replace:
_raise QschReadingError(f"Failed to find the net for {pin} in component in position {comp_pos}")_ with raise _QschReadingError(f"Failed to find the net for {pin} in component in position {comp_pos} schematic {self._qsch_file_path}") in qsch_editor._

See example attached where one port is not connected.
[spicelib Qspice bug floating net subcircuit.zip](https://github.com/user-attachments/files/17001934/spicelib.Qspice.bug.floating.net.subcircuit.zip)

hb020, 2024-09-14:
Will correct and include this case in the test suite

nunobrum, 2024-09-14:
I am fixing this in the https://github.com/nunobrum/spicelib/tree/qsch_all_elem branch

hb020, 2024-09-14:
ok, I just corrected it (pushed that into branch "issue88-hb" , 

but found another issue: the netlist file that is created is as follows:

```
* ~/Documents/workspace/spicelib/examples/testfiles/Qspice_bug_floating_net.qsch
V1 N01 0 48
R1 N01 N02 1µ
L1 N02 Vin 5µ
C1 Vin 0 600µ
C2 N03 0 100µ
R4 Vout 0 80m
R5 Vout N03 1m
R7 N02 Vin 1
X1 Vin Vout Qspice_bug_floating_net_X1

.subckt Qspice_bug_floating_net_X1 input output some_measurement
R2 input N01 10µ
L2 N02 output 10µ
R6 some_measurement N02 1m
V2 N03 0 pulse 0 1 0 10n 10n 2.5µ 10µ
D1 0 some_measurement D
D2 some_measurement N01 D
R1 N01 some_measurement 1
.model SW SW(Ron=1Âµ Roff=1Meg Vt=.5 Vh=-.4)
.lib Diode.txt
.ends Qspice_bug_floating_net_X1

.tran 100Âµ 10m 10Âµ 100n
.plot V(Vout)
.end
```
This is in UTF-8
Look at the "Âµ" and "µ" The first form is from the main file, the latter from the subcircuit.

"Âµ" is the form of "µ" when the source was in utf-8 but read in cp1252.

nunobrum, 2024-09-14:
I see the error now that I'm trying to push my "qsch_all_elem" branch. It doesn't happen in Windows, but, with your tip I think I understand where is the problem.

By the way, I don't find the "issue88-hb". Did you push it ?

hb020, 2024-09-14:
It's there: https://github.com/nunobrum/spicelib/tree/issue88-hb

nunobrum, 2024-09-14:
Thanks @hb020 . 
In Windows the example from @TorfromNor works well with the my branch, even with the Â character. QSPICE ignores it.
However, I think that it is very weird that inductances don't have the added character. Only the texts. Maybe they have a different treatment. Something to look closer.

I'm seeing your branch.  I like you added the file to the testbench.  I'll merge the two branches, before merging to main.

nunobrum, 2024-09-14:
Fixed in PR #90 (V1.2.1)

closed 2024-09-14 by nunobrum

TorfromNor, 2024-09-15:
> Thanks @hb020 . In Windows the example from @TorfromNor works well with the my branch, even with the Â character. QSPICE ignores it. However, I think that it is very weird that inductances don't [...]

I checked this issue is present inside Qspice itself maybe it is a windows region / language setting  or similar which lead to this version. 

[image]

nunobrum, 2024-09-15:
Interesting observation @TorfromNor. I guess QSPICE is far from being stable. 
When looking at the netlist that is generated by QSPICE, I see things that really puzzle me. 
Here are a few examples:

`×1 «N59 0 N58 0» Turns=1 1` 
Why are the this element ports within «»

`Q3 0 N39 N38 [0] ZXTP2027F PNP`
Why is the extra pin added between []

I coded the qsch_editor.py so it would produce a similar netlist, but, I wonder....

TorfromNor, 2024-09-15:
> Interesting observation @TorfromNor. I guess QSPICE is far from being stable. When looking at the netlist that is generated by QSPICE, I see things that really puzzle me. Here are a few examples: [...]

It could be the << >> relates to verilog/ccode blocks atleast when i open one of the basic exampels i made with Verilog and Ccode i see:

Ø´X1 «N01´f» «N02´f» «»  Ccode
V1 N01 0 SINE 0 1 1K
Ø´X2 «N01´f» «N03´f» «»  Verilog
[image]

There is allot of block types in Qspice which might have some strange handling:
[image]