nunobrum/spicelib #101: Changing values of internal components is handled incorrectly with AscEditor

hb020, 2024-09-29:
Suppose a .asc file that refers to a component U1 that is inside a library.

It is perfectly possible to do `my_asc_editor["U1:R1"].value`. Why not.
It is however also possible to do `my_asc_editor["U1:R1"].value = 1`
BUT: when calling `my_asc_editor.save_netlist()`, that change is lost, as the library is not rewritten (and that would even be dangerous)

The same is not a problem with netlists (SpiceEditor), as it creates a copy of the component upon changes.

Therefore: this is not as expected.

3 options:

1.  make direct library internal component value changes impossible
2.  rewrite the library upon `my_asc_editor.save_netlist()`
3.  force the component value via model change directives in Value or Value2

Not sure what the best way is. Nr 1 is the easiest probably, and still leaves the user the option for nr 3.

nunobrum, 2024-09-29:
I understand what you mean. 
The current behaviour of AscEditor is inconsistent with what was implemented for the SpiceEditor and this is confusing. The big difference is that adding N instances of a .SUBCKT is less costly than to store N instances of .ASC files.

I agree that option 1. is the easiest and probably the one that makes most sense. 
The option 2. would imply creating a new library and dump there all library changes. In case of BLOCKs even going to the point of converting the .ASC into a SpiceNetlist. I think it is a lot of work for something that is probably not going to be very used.

The current implementation was brought up by the idea that  if your netlist is
```
U1 A B C MySubCircuit
U2 D E F MySubCircuit
```
when writing my_asc_editor["U1:R1"].value = 1, the U2 should not be updated, as it **might not be** what the user wants.

However, I guess that **most users would prefer** the update of ALL instances of a sub-circuit rather than just one. 
BTW, one of the latest additions to the library the `get_subcircuit_named("MySubCircuit")` had the purpose of allowing the user to do just that. That is, option 3. is always possible, and it is possible to do it without having to manipulate Value and Value2 directly.

Maybe the error at the beginning was to implement that modified_subcircuits approach. Should we abandon it?

hb020, 2024-09-29:
> Maybe the error at the beginning was to implement that modified_subcircuits approach. Should we abandon it?

Nah, it is in there now, works pretty OK as far as I can see. No need to remove it. 
No guesses on what the users want (all instances or individual). 
LTspice doesn't allow easy library changes either. 

For now, the most robust would be to just forbid changes to components from libraries. So: nr 1.

If you want to allow changes to libraries, a new type of editor might be needed. And for how many users?

nunobrum, 2024-09-29:
> Nah, it is in there now, works pretty OK as far as I can see. No need to remove it.
That's fair. What is done is done, it does work well and now being properly tested.

When I started to code the object-oriented approach that allows the netllist[ref].value, the initial code was moving most of the code that is today in the Editor classes towards the Component Class. That would a much cleaner code, and in a long run, would allow to move away from regular expressions, with each component receiving it's own class. Except for two things, 1) it broke the modified subcircuit implementation, and 2) the time it was taking and prospect of porting the same concept toward all the other Editors was going to be too much time consuming. Quite frankly I'm struggling to find time for this.

So, I'm happy if you take the time to work on 1.  And honestly, I don't know how many users are interested in changing libraries. Maybe less than 1%. I mean... Keeping validated and trustworthy code is the purpose of libraries.
Applying this principle to spicelib, it is better to reduce the functionality to the bare minimum and make sure that everything works as expected. 
Thanks again @hb020 for your contributions.

hb020, 2024-10-01:
we should document this, probably with the hierarchy stuff.

closed 2024-10-07 by hb020