nunobrum/spicelib #109: [LTSpice] how to give L,W attribute to NMOS,PMOS

optimizerer, 2024-10-11:
Hi, spicelib developers and users.
I try to trace ASCO(one optimizer for spice) example of 3-stage opamp by LTSpice.
https://asco.sourceforge.net/examples.html
  (intend to apply my optimization software named "OptParaSol" )

I need to change MOS attribute(?) W,L (gate Width, Length) by each simulation.
Please let me know How to point W,L.
  (In case normal register/capasitor value,  
   Those can be done by spicelib like
   netlist.set_component_value('R1','2k') ) 

[image]

hb020, 2024-10-12:
Indeed, it a missing feature at present, parameter substitution is not yet not supported for MOS (nor for Bipolar or Lossy Transition lines). Will add it.

@nunobrum : also, when the subcircuit name contains a `.`, like in ```XOPAMP VP VN VOUT IBIAS VB1 VDD VSS PFC.SUB```, it breaks. Worth a repair as well?

hb020, 2024-10-12:
[amp3.net.txt](https://github.com/user-attachments/files/17350902/amp3.net.txt)
This is my test file. Just rename it to .net. I removed the `.SUB` from the subcircuit definition and invocation.

hb020, 2024-10-12:
See branch [Allow-MOS-parameters-#109](https://github.com/nunobrum/spicelib/tree/Allow-MOS-parameters-%23109)

How to use it:
```python
netlist = spicelib.SpiceEditor("amp3.net")
print("Before")
l = netlist.get_subcircuit("XOPAMP").get_component_parameters('M00').items()
print(f"XOPAMP:M00 params={l}")
newsettings = {"W": 10E-6}
netlist.get_subcircuit("XOPAMP").set_component_parameters("M00", **newsettings) 
# or: netlist.get_subcircuit("XOPAMP").set_component_parameters("M00", W=10E-6)
print("After")
l = netlist.get_subcircuit("XOPAMP").get_component_parameters('M00').items()
print(f"XOPAMP:M00 params={l}")
```

Note that for now, you must avoid using a dot in a subcircuit name, see the example file.

nunobrum, 2024-10-12:
I'll also repair the issue with the dot in the subcircuit,

nunobrum, 2024-10-13:
Fix included in 1.3.1

closed 2024-10-13 by nunobrum