nunobrum/spicelib #115: ReadRaw function bytes error

jiaxinmax, 2024-12-03:
The raw file generated by ngspice in Ubuntu is similar to that generated by ltspice in win.If the raw file of ngspice contains the v-sweep variable, the ReadRaw function will report the error in calculating the block size. Expected 56 bytes, but found 57 bytes.

Since the RAW file cannot be uploaded, here the extension has been changed to txt

[1.txt](https://github.com/user-attachments/files/17986056/1.txt)

hb020, 2024-12-03:
Are you sure the attached file is the file that provokes the error? An error message "Expected xx bytes, but found yy bytes" will only occur of you have a raw file with binary values, but the attached file has text values. 
See the line "Values:" if that was absent, and the line would say "Binary:", then we activate the code that might show that error message.

The attached file works OK on my test systems (MacOS and Ubuntu)

jiaxinmax, 2024-12-05:
> 您确定附件是引发错误的文件吗？仅当您的原始文件包含二进制值，而附件文本包含值时，才会出现错误消息“预期为 xx 字节，但发现为 yy 字节”。 请参阅“值：”行。如果没有该行，并且该行显示“二进制：”，则我们激活可能显示该错误消息的代码。
> 
> 也许还有压缩包？
> 
> 附件在我的测试系统（MacOS 和 Ubuntu）上运行良好

sorry,The file I sent before was successfully run under win.The following is a file with a bug in Ubuntu,thanks!

[bug.zip](https://github.com/user-attachments/files/18016375/bug.zip)

hb020, 2024-12-05:
This is more of a definition problem. The .cir file requests 2 data sets: DC and TRAN. 

The result is that the raw file has 2 data sets in the same file. This is the case for both the raw files (the text file and the binary file): They have 2 sets of data, concatenated in the same file. Just look with a text editor or a hex editor, and you see that that both files have a first set with 
```
Plotname: DC transfer characteristic
Flags: real
No. Variables: 9
No. Points: 101     
Variables:
	0	v(v-sweep)	voltage
	1	v(6)	voltage
	2	v(1)	voltage
	3	v(2)	voltage
	4	v(3)	voltage
	5	v(4)	voltage
	6	v(5)	voltage
	7	i(vin)	current
	8	i(vcc)	current
```
and a second set with
```
Plotname: Transient Analysis
Flags: real
No. Variables: 9
No. Points: 122     
Variables:
	0	time	time
	1	v(6)	voltage
	2	v(1)	voltage
	3	v(2)	voltage
	4	v(3)	voltage
	5	v(4)	voltage
	6	v(5)	voltage
	7	i(vin)	current
	8	i(vcc)	current
```

If you separate the 2 data sets, they work in spicelib.

It would be best to produce only 1 data set from the circuit file.

jiaxinmax, 2024-12-07:
> This is more of a definition problem. The .cir file requests 2 data sets: DC and TRAN.
> 
> The result is that the raw file has 2 data sets in the same file. This is the case for both the raw files (the text file and the binary file): They have 2 sets of data, concatenated in the same file. Just look with a text editor or a hex editor, and you see that both files have a first set with
> 
> ```
> Plotname: DC transfer characteristic
> Flags: real
> No. Variables: 9
> No. Points: 101     
> Variables:
> 	0	v(v-sweep)	voltage
> 	1	v(6)	voltage
> 	2	v(1)	voltage
> 	3	v(2)	voltage
> 	4	v(3)	voltage
> 	5	v(4)	voltage
> 	6	v(5)	voltage
> 	7	i(vin)	current
> 	8	i(vcc)	current
> ```
> 
> and a second set with
> 
> ```
> Plotname: Transient Analysis
> Flags: real
> No. Variables: 9
> No. Points: 122     
> Variables:
> 	0	time	time
> 	1	v(6)	voltage
> 	2	v(1)	voltage
> 	3	v(2)	voltage
> 	4	v(3)	voltage
> 	5	v(4)	voltage
> 	6	v(5)	voltage
> 	7	i(vin)	current
> 	8	i(vcc)	current
> ```
> 
> If you separate the 2 data sets, they work in spicelib.
> 
> It would be best to produce only 1 data set from the circuit file. See attached file for the file, split. [Archive.zip](https://github.com/user-attachments/files/18021846/Archive.zip)
> 
> The reason that the text version worked in spicelib, is that spicelib only picks up the first set. ngspice itself strangely enough only picks up the last set.

Thank you very much for your response.

closed 2024-12-07 by hb020