nunobrum/spicelib #120: LTspice: fraprobe.asy issue &xxx

TorfromNor, 2025-01-07:
Hello,
I get the following error when the schematic contains the LTspice fraprobe. (Which is somewhat newly added)

"
NotImplementedError: Primitive not supported for ASY file
"Version 4
"

The special thing with FRA probe is the & symbol type however the file from LTspice seem a bit wierd also i dont think the "Version 4 should normally activate this error since it is not the relevant line. In addition the error message show allot of spaces on this line which seem a bit wierd. (tried a few things with the "elif line.startswith("Version"):" in ASY_reader but no luck)

Other things i tried:
- I tried to open the fraprobe.asy file in notepad and resave,  that broke the file which is a bit strange since it dont happen with other .asy files. 

So i suspect encoding issue and something related to the symbol type &

See attached for example files. 
[09 spicelib LTspice FRA probe issue.zip](https://github.com/user-attachments/files/18333596/09.spicelib.LTspice.FRA.probe.issue.zip)

hb020, 2025-01-07:
Confirmed, this is a unicode problem. Will propose a patch soon.

TorfromNor, 2025-01-07:
Thanks so it was in the asy reader the encoding issue was. 

Regarding the trace it was a mistake and not related to spicelib. I tested first with the other file: "01 Opamp bode open loop - FRA Probe removed" which worked.

The FRA probe changes the simulation a bit so gain_1 can be shorted to ground. Adding a minor resitance from FRA to ground make gain_1 come back also with the probe.

nunobrum, 2025-01-08:
Hello all,

I'm happy that this one was fixed quicker than I had time to address it. Thanks @hb020 once again.
I have merged the PR.
Can this be closed ?

Cheers,
Nuno

closed 2025-01-11 by nunobrum