nunobrum/spicelib #122: Qspice - sub circuit bug with named components. (netlist generated with unkown mistake)

TorfromNor, 2025-01-10:
Hello,
I got this error:
ERROR:spicelib.RunTask:RunTask #1:: Simulation Aborted. Time elapsed: 00.0025 secs
ERROR:spicelib.SimRunner:Skipping 1 because simulation failed.

This issue is due to the .net file generated by spicelib for Qspice is not able to handle sub circuits with components with custom names. See attached test files.

**What i found out:**
- spicelib can handle custom component names on top schematic (for example renaming R1 to current_shunt)
- spicelib can handle custom component names in sub circuits (for example renaming R1 to {sub_assembly_name}_measurement)
- **spicelib can not handle custom component name in sub circuit if they are just named measurement (missing the sub assembly name)**

I suspect spicelib malforms the netlist, when the component name dont match the sub circuit name, however looking into the netlist generated, it is not obvius to me what goes wrong. (and spicelib gives no error or indication of a problem since the netlist is generated successfully)

**Netlist inspection**
--------------------------------------------------------------------------------
**Working netlist with named sub component similar as sub circuit**
V1 N01 0 48
R1 N01 N03 1µ
R4 N02 0 80m
X1 N03 N02 Current_shunt

.subckt Current_shunt in out
Current_Shunt_measurement in out 1m
.ends Current_shunt

.tran 100Âµ 10m 10Âµ 100n
.end

--------------------------------------------------------------------------------
**Not working netlist with different naming in sub circuit**
Part of the not working 
X1 N03 N02 Current_shunt_different_name

.subckt Current_shunt_different_name in out
measurement in out 1m
.ends Current_shunt_different_name

In both cases the subckt generated by spicelib seems similar and roughly like i would have expected (without being to familiar with the netlist definitions) however qspice is not able to use the last netlist (qspice is able to run all the base test files)

--------------------------------------------------------------------------------
**Test files:**
[10 Qspice bug with sub circuit loading.zip](https://github.com/user-attachments/files/18375195/10.Qspice.bug.with.sub.circuit.loading.zip)

nunobrum, 2025-01-11:
Hi @TorfromNor 

the subcircuits have a different treatment in regards to the other components. 
It is so to be able to make edits on a single instance of a circuit.  This is a behavior that was difficult to explain.

But honestly I am very confused with this issue. The description doesn't really match the title and the zip file you attached to it.

Can you please revise this?

Best,
N.

TorfromNor, 2025-01-13:
Hi @nunobrum Sorry about that thought they fitted however i see i also forgot to clean up the folders. I have understood the issue much better now and updated the description and explainations + test files. 

The issue is also not only related to subcircuits but any netlist generated. Hopefully it is more clear now that the netlist output from spice lib is missing in some cases a marker for what the component type is.

A part of the issue is also that spicelib gives no errors, Qspice can often run the netlist however will give completly wrong simulation results due to Qspice using the wrong component type in certain circumstances.

nunobrum, 2025-01-19:
Hi @TorfromNor ,

Can you give a bit more background. Give another zip where I can verify the failure.

Thanks in advance,
Nuno

TorfromNor, 2025-01-28:
Hi @nunobrum,
See attached for more examples of errors in Qspice netlists. 

[spicelib_Qspice_wrong_netlists_output.zip](https://github.com/user-attachments/files/18572864/spicelib_Qspice_wrong_netlists_output.zip)

As a side note:
LTspice netlists behaves similar for such named components but uses a other symbol, Qspice uses ´ while LTspice netlists uses § see bellow for a example for LTspice. 

R§current_shunt N001 0 9m
V1 N002 0 PULSE(0 1 500u 10n 10n 1)
R1 N001 N002 1m
.tran 1m
.backanno
.end

I think spicelib mainly uses the ASC files directly for LTspice simulations and then the issue is not present however in cases with netlist usage could see such problems.

nunobrum, 2025-01-30:
Hi @TorfromNor ,

only now I had the time to download and run your code. 
Thanks for the effort you placed in explaining the issue. 
I need a bit more time to fix this. Having elements with names starting with an opposite letter is not 
something that I was expecting to have.

Thanks for your patience,
Nuno

nunobrum, 2025-02-08:
Going to be included in 1.3.7

closed 2025-02-08 by nunobrum