nunobrum/spicelib #126: subcircuit in given folder

bam1rt, 2025-01-28:
Dear Nunobrum

I have an .asy file using subcircuits (.asy) in the same folder and it does not work.  I traced it down and think this is the reason; none of of the attributes `('ModelFile', 'SpiceModel', 'SpiceLine', 'SpiceLine2', 'Def_Sub', 'Value', 'Value2')` are used and it works fine within LTspice. The path is also set in `self._asy_file_path` so could you just return this variable instead if `self.attributes.get('SpiceModel')` ?

BR, BastelBaus

hb020, 2025-01-28:
Hi, do you have a (simplified) example that we can test on?

bam1rt, 2025-02-21:
So, finally I learnt some more;
- I am using a subcircuit as block from an .asc / .asy file combination
- both files are in another subdirectory than the main file
- the location of the .asy (symbol) file is in my LTSpice settings symbol seach path
- The symbol is found by PyLTSPice but not the circuit (.asc file)
- It works nice in LTSpice itself 
Testcase attached:
[testcase.zip](https://github.com/user-attachments/files/18916374/testcase.zip)

My fix was adding both lines before the return statement of get model https://github.com/nunobrum/spicelib/blob/dd8d9c5b145507ae17780483b37231408a149e96/spicelib/editor/asy_reader.py#L288:
        guess_file = self._asy_file_path.with_suffix('.asc')
        if guess_file.is_file(): return str(guess_file.resolve().as_posix())

Alternativ one could use the default search paths also for subcircuits but I am not so deep in the code

nunobrum, 2025-02-22:
Hello @bam1rt,

I am a bit lost with this issue When I open the file in my LTspice I see this.

[image]
I have no idea how it works on your computer. In my installation, I need to place the subcircuit and its symbol on the same directly for the LTspice to work. 
This is what spicelib is expecting.

bam1rt, 2025-02-22:
You have to add the the sub-folder into the default search path, then it works well in ltspice but not in PyLTSPice:

closed 2025-02-22 by bam1rt