nunobrum/spicelib #127: `.param` Value Replacement Leads to Duplicated Numeric Values (e.g., `0.05` → `0.05.05`)

womo75, 2025-01-29:
Hi, I'm encountering an issue in this library when setting a parameter value via `set_parameter()`. Specifically, if I have a line in my SPICE netlist such as:
```
.param Vamp=0.05
```
and I call:
```python
netlist.set_parameter("Vamp", 0.05)
```
the resulting line becomes:
```
.param Vamp=50m.05
```
instead of the expected:
```
.param Vamp=50m
```
(where `format_eng(0.05)` results in `50m`, which is correct, but then the original `.05` is left over in the text).

It appears that the regular expression or matching logic for extracting and replacing the parameter's value is inadvertently capturing part of the original string (including the `=` or part of the number) so that the new value is appended without removing the old one properly.

---

**Steps to Reproduce:**
1. In a SPICE netlist, include a line with a floating-point parameter, for example:  
   ```
   .param Vamp=0.05
   ```
2. Use the following Python code:
   ```python
   netlist.set_parameter("Vamp", 0.05)
   ```
3. Check the modified netlist. The `.param` line is transformed into:
   ```
   .param Vamp=50m.05
   ```
   instead of a proper `50m` replacement.

---

**Expected Behavior:**  
- The `.param Vamp=0.05` line should become `.param Vamp=50m` (or `.param Vamp=0.05` if passing a string rather than a float).  
- In other words, the old value (including any trailing characters) should be fully replaced by the new value.

---

**Actual Behavior:**  
- The new value is appended to the old floating number, resulting in `.param Vamp=50m.05`.

---

**Additional Observations / Potential Cause:**  
- The `_get_param_named()` method or an equivalent function might include the `=` in the regex capture or incorrectly determine the `start`/`end` indices for the value. As a result, the code that performs `line[:start] + value_str + line[end:]` doesn’t remove the whole substring for the parameter’s old value.  
- A potential fix is to ensure that the regex capturing group for the value does not include the `=` sign and that it captures only the numeric/string part that should be replaced, thereby avoiding leftover text.

---

**Suggested Fix:**  
- Adjust the regular expression (or whichever logic identifies the parameter value in `.param XXX=YYY`) so that only the value `YYY` is captured and replaced.  
- For instance, something like:
  ```regex
  ^[ \t]*\.param[ \t]+(?P<name>\w+)[ \t]*=[ \t]*(?P<value>\S+)
  ```
  so that the capture group `value` starts right after `=` and whitespace, ensuring no trailing original text remains after the replacement.

Thank you for looking into this issue!

nunobrum, 2025-01-29:
I just had a quick look into it. I think the problem is that I forgot to add the '.' into the value regular expression.
I think that updating this function on the base_editor.py as so will fix the problem

```
def PARAM_REGEX(pname):
    return r"(?P<name>" + pname + r")\s*[= ]\s*(?P<value>(?P<cb>\{)?(?(cb)[^\}]*\}|[\d\.\+\-Ee]+[a-zA-Z%]*))"
```

nunobrum, 2025-01-30:
Fix included in 1.3.6. 
I am confident that it will fix your problem.

closed 2025-01-30 by nunobrum