nunobrum/spicelib #129: Error on Loading a Hierarchical Schematic in `.qsch` Format: `IndexError: list index out of range`

womo75, 2025-02-01:
**Description:**  
When attempting to load a hierarchical schematic file (with sub-circuits) using `QschEditor(cir_name + ".qsch")`, I get the following error trace:

```
IndexError                                Traceback (most recent call last)
Cell In[23], line 1
----> 1 QschEditor(cir_name + ".qsch")

File ...\qsch_editor.py:358, in QschEditor.__init__(self, qsch_file, create_blank)
    ...
File ...\qsch_editor.py:573, in QschEditor.reset_netlist(self, create_blank)
    ...
File ...\qsch_editor.py:628, in QschEditor._parse_qsch_stream(self, stream)
    ...
File ...\qsch_editor.py:325, in QschTag.get_text(self, label, default)
    ...
IndexError: list index out of range
```

This error occurs specifically when the `.qsch` file contains one or more sub-circuits (hierarchical design). It seems to break during `_parse_qsch_stream`, at the line:

```
sch_comp.attributes['type'] = symbol.get_text('type', "X")
```

Thank you for taking a look at this issue!

nunobrum, 2025-02-01:
Hello @womo75 ,

can you share the offending .qsch file.
That would help pin-point the problem.

Best Regards,
Nuno

womo75, 2025-02-07:
[test.zip](https://github.com/user-attachments/files/18709662/test.zip)

Thank you for your response.

I have attached the .qsch file that may be causing the issue. Please take a look.

Best regards,

closed 2025-02-08 by nunobrum