nunobrum/spicelib #130: QspiceLogReader: Missing .net file in QSPICE environment – Use .cir file instead

womo75, 2025-02-01:
Description:
In our QSPICE environment, the simulation produces a .cir file but does not generate the corresponding .net file.
Since the QspiceLogReader class in spicelib assumes the existence of a file with the same base name but a .net extension , a FileNotFoundError occurs when attempting to parse the measurement results.

To work around this issue, I modified the obtain_measures() method in qspice_log_reader.py so that it uses the .cir file instead of the .net file. This allows QPOST to generate the .meas file correctly, and the measurement data can then be parsed without errors.

nunobrum, 2025-02-01:
I'm taking note of this and will make sure that next version will try to open a .cir if it doesn't find a .net.

hb020, 2025-02-07:
Found another bug that function. @nunobrum I'll leave it up to you to review and maybe add the other qspice stuff. My qspice install is broken now, cannot test.

nunobrum, 2025-02-08:
I am seeing your last message from yesterday now after I have merged. 
I'll check on my side and and make a correcting branch if needed.

nunobrum, 2025-02-08:
All seems to be OK. Closing this issue

closed 2025-02-08 by nunobrum