nunobrum/spicelib #131: AssertionError: Unsupported version : 4.1

gudnimg, 2025-02-03:
I suspect the issue is with newer LTspice version was recently released.

[image]

```
---------------------------------------------------------------------------
AssertionError                            Traceback (most recent call last)
Cell In[1], [line 10](vscode-notebook-cell:?execution_count=1&line=10)
      [7](vscode-notebook-cell:?execution_count=1&line=7) from spicelib import AscEditor, SimRunner  # Imports the class that manipulates the asc file
      [9](vscode-notebook-cell:?execution_count=1&line=9) # Load the circuit drawing and generate a net list file
---> [10](vscode-notebook-cell:?execution_count=1&line=10) netlist = AscEditor("directory/mydrawing.asc")
     [12](vscode-notebook-cell:?execution_count=1&line=12) threads_count = psutil.cpu_count()
     [13](vscode-notebook-cell:?execution_count=1&line=13) print(f"Detected {threads_count} CPU threads")

File c:\Users\gudnimar\AppData\Local\Programs\Python\Python313\Lib\site-packages\spicelib\editor\asc_editor.py:75, in AscEditor.__init__(self, asc_file, encoding)
     [73](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:73)     self.encoding = encoding  
     [74](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:74) # read the file into memory
---> [75](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:75) self.reset_netlist()

File c:\Users\gudnimar\AppData\Local\Programs\Python\Python313\Lib\site-packages\spicelib\editor\asc_editor.py:209, in AscEditor.reset_netlist(self, create_blank)
    [207](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:207) elif line.startswith("Version"):
    [208](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:208)     tag, version = line.split()
--> [209](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:209)     assert version in ["4"], f"Unsupported version : {version}"
    [210](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:210)     self.version = version
    [211](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:211) elif line.startswith("SHEET "):

AssertionError: Unsupported version : 4.1
```

[image]

hb020, 2025-02-03:
See PR #135

nunobrum, 2025-02-03:
Merged the PR #135 . Will close when this is packaged to PyPi

nunobrum, 2025-02-08:
Included in 1.3.7

closed 2025-02-08 by nunobrum