nunobrum/spicelib #132: .asy file incorrectly parsed as .asc

gudnimg, 2025-02-03:
I have attached to this issue a minimal drawing to reproduce the issue. My .asy symbol is just a simple transistor.

* Python 3.13.1
* Spicelib 1.3.6
* LTspice project to reproduce  [reproducing_error.zip](https://github.com/user-attachments/files/18639542/reproducing_error.zip)

The code to reproduce this:

```python
from spicelib import AscEditor, SimRunner  # Imports the class that manipulates the asc file

netlist = AscEditor("reproducing_error/drawing.asc")
```

```
---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
Cell In[1], [line 10](vscode-notebook-cell:?execution_count=1&line=10)
      [7](vscode-notebook-cell:?execution_count=1&line=7) from spicelib import AscEditor, SimRunner  # Imports the class that manipulates the asc file
      [9](vscode-notebook-cell:?execution_count=1&line=9) # Load the circuit drawing and generate a net list file
---> [10](vscode-notebook-cell:?execution_count=1&line=10) netlist = AscEditor("reproducing_error/drawing.asc")
     [12](vscode-notebook-cell:?execution_count=1&line=12) threads_count = psutil.cpu_count()
     [13](vscode-notebook-cell:?execution_count=1&line=13) print(f"Detected {threads_count} CPU threads")

File c:\Users\gudnimar\AppData\Local\Programs\Python\Python313\Lib\site-packages\spicelib\editor\asc_editor.py:75, in AscEditor.__init__(self, asc_file, encoding)
     [73](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:73)     self.encoding = encoding  
     [74](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:74) # read the file into memory
---> [75](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:75) self.reset_netlist()

File c:\Users\gudnimar\AppData\Local\Programs\Python\Python313\Lib\site-packages\spicelib\editor\asc_editor.py:173, in AscEditor.reset_netlist(self, create_blank)
    [170](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:170)     symbol = self._get_symbol(component.symbol)
    [171](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:171)     if component.reference.startswith('X') or symbol.is_subcircuit():  # This is a subcircuit
    [172](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:172)         # then create the attribute "SUBCKT"
--> [173](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:173)         component.attributes['_SUBCKT'] = self._get_subcircuit(symbol)
    [174](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:174) else:
    [175](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:175)     # make sure prefix is uppercase, as this is used in a lot of places
    [176](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:176)     if ref.upper() == "PREFIX":

File c:\Users\gudnimar\AppData\Local\Programs\Python\Python313\Lib\site-packages\spicelib\editor\asc_editor.py:290, in AscEditor._get_subcircuit(self, symbol)
    [284](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:284)         asc_path = search_file_in_containers(asc_filename.stem + os.path.extsep + "asc",  # file to search
    [285](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:285)                                              os.path.split(self.asc_file_path)[0],  # The current script directory
    [286](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:286)                                              os.path.curdir,  # The directory where the script is located
    [287](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:287)                                              *self.custom_lib_paths  # The custom library paths. They are last here, contrary to other places... Why?
    [288](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:288)                                              )
    [289](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:289)     if asc_path is None:
--> [290](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:290)         raise FileNotFoundError(f"File {asc_filename} not found")
    [291](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:291)     answer = AscEditor(asc_path)
    [292](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:292) elif lib is None and symbol.symbol_type == "CELL":
    [293](file:///C:/Users/gudnimar/AppData/Local/Programs/Python/Python313/Lib/site-packages/spicelib/editor/asc_editor.py:293)     # TODO: the library is often specified later on, so this may need to move.

FileNotFoundError: File reproducing_error\DMN61D9UDWQ.asc not found
```

gudnimg, 2025-02-03:
Running the simulation in LTspice works fine.

gudnimg, 2025-02-03:
This seems to fix the error. However, when I run a simulation the component is not added properly in the .net file 🤔 

[image]

hb020, 2025-02-03:
Yes, it is a bit more complicated. 

> This seems to fix the error. However, when I run a simulation the component is not added properly in the .net file 🤔
> 
> [image]

This is likely to provoke other errors indeed.

gudnimg, 2025-02-03:
Adding the symbol/library to LTspice's search path seems to resolve the issue. I don't recall having needed to do this before though 🤔 . I may be wrong though.

[image]

hb020, 2025-02-03:
That is indeed the approach I was looking for. Including it in the library search. And you can adapt spicelib to expand the lib search path.

gudnimg, 2025-02-03:
>And you can adapt spicelib to expand the lib search path.

I will add that to my script. That way anyone opening my scripts won't need to open LTspice to explicitly set these search paths.

So I guess the only issue here is to handle `.txt` library files. Not sure how common that is, but this is how Diodes Incorporated are supplying the library/model.

hb020, 2025-02-03:
Just did a commit, see PR#136 (well, you'd need the others PRs as well)

gudnimg, 2025-02-03:
@hb020 Thanks for taking the time to look into this!

hb020, 2025-02-03:
The version you had is nicer. Just committed that version.

nunobrum, 2025-02-03:
Merged pull-request. Will close the issue once the modification crawls its way to the PyPi

closed 2025-02-08 by nunobrum