nunobrum/spicelib #137: FileNotFoundError: File voltage.asy not found

NooriDan, 2025-02-03:
Hi, I'm trying to run the following LTSpice file in python using 

`from spicelib import AscEditor`
`netlist = AscEditor("asc/LPF_3rd.asc") `

but I get the following error. I have LTSpice installed on my Mac and I can run everything in LTSpice GUI smoothly: 

---------------------------------------------------------------------------
FileNotFoundError                         Traceback (most recent call last)
Cell In[37], line 1
----> 1 netlist = AscEditor("asc/LPF_3rd.asc")  # Open the Spice Model, and creates the .net
      2 netlist

File ~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/editor/asc_editor.py:75, in AscEditor.__init__(self, asc_file, encoding)
     73     self.encoding = encoding  
     74 # read the file into memory
---> 75 self.reset_netlist()

File ~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/editor/asc_editor.py:170, in AscEditor.reset_netlist(self, create_blank)
    168 if ref == "InstName":
    169     component.reference = text
--> 170     symbol = self._get_symbol(component.symbol)
    171     if component.reference.startswith('X') or symbol.is_subcircuit():  # This is a subcircuit
    172         # then create the attribute "SUBCKT"
    173         component.attributes['_SUBCKT'] = self._get_subcircuit(symbol)

File ~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/editor/asc_editor.py:265, in AscEditor._get_symbol(self, symbol)
    263 asy_path = self._asy_file_find(asy_filename)
    264 if asy_path is None:
--> 265     raise FileNotFoundError(f"File {asy_filename} not found")
    266 answer = AsyReader(asy_path)
    267 return answer

FileNotFoundError: File voltage.asy not found

hb020, 2025-02-03:
Can you show the output of the following?
```
from spicelib.simulators.ltspice_simulator import LTspice

print(LTspice.spice_exe)
print(LTspice.get_default_library_paths())
```

(If you want, you can just use 'myuser' instead of your real user name if your user name is in the path)

The `voltage.asy` file should be in one of the paths from the last print statement, as `sim/voltage.asy`

timo-braungardt, 2025-02-04:
Just experienced a similar issue. The path `~/Documents/LtspiceXVII/lib/` in `spicelib/simulators/ltspice_simulator.py` needs a capitalized `LTspiceXVII`.

NooriDan, 2025-02-04:
> Can you show the output of the following?
> 
> ```
> from spicelib.simulators.ltspice_simulator import LTspice
> 
> print(LTspice.spice_exe)
> print(LTspice.get_default_library_paths())
> ```
> 
> (If you want, you can just use 'myuser' instead of your real user name if your user name is in the path)
> 
> You should get something like
> 
> ```
> ['/Applications/LTspice.app/Contents/MacOS/LTspice']
> ['/Users/myuser/Documents/LtspiceXVII/lib/']
> ```
> 
> The `voltage.asy` file should be in the path (or one of the paths) from the last print statement, as `sim/voltage.asy`

I get 

['/Applications/LTspice.app/Contents/MacOS/LTspice']
[]

I think something has gone wrong with finding the lib files on macos?

hb020, 2025-02-04:
> Just experienced a similar issue. The path `~/Documents/LtspiceXVII/lib/` in `spicelib/simulators/ltspice_simulator.py` needs a capitalized `LTspiceXVII`.

Indeed, it might be that, See branch [FileNotFoundError-on-MacOS](https://github.com/nunobrum/spicelib/tree/FileNotFoundError-on-MacOS).

If that doesn't solve it, please do a search for 'voltage.asy' with finder and tell me what path it is in. In the meantime, you can use `AscEditor.set_custom_library_paths()`, but of course I will try to make this work on all systems.

You probably use case sensitive file systems. Not common on MacOS. I suspect that you'll have issues with library names. If so, I will need to add extra measures.

timo-braungardt, 2025-02-04:
The fix and the workarround work for me.

But I might add that im not on MacOS but Ubuntu with wine.
So apparently wine uses 'My Documents' instead of 'Documents' and I had to additionally change that. 
You could add `~/My Documents/LTspiceXVII/lib/` to the `_default_lib_paths` array. (already tested it, worked)

Thanks for the fast response! I hope it also works on MacOS now.

hb020, 2025-02-04:
Indeed, "Documents" or "My Documents" varies. I added the path in the branch. 
Waiting for feedback from @NooriDan before validating.

NooriDan, 2025-02-04:
I still cant find ~/Documents/LTspiceXVII/lib/. In fact, I only see an LTspice folder in my Documents (not LTspiceXVII). I tried overwriting the default path but its not working either. I append the new path to  runner.simulator._default_lib_paths and when I call print(runner.simulator.get_default_library_paths()) it still returns an empty list

NooriDan, 2025-02-04:
So I found the lib folder in "/Applications/LTspice.app/Contents/Resources" as lib.zip. and when I unzip it i see three folders cmp, sub, and sym, where I can see voltage.asy. I extract the zip and place it in "/Users/username/Documents/LTspice", and then tried setting it by calling `AscEditor.set_custom_library_paths(['~/Documents/LTspice/lib'])`. I see it is then added to AscEditor.custom_lib_paths but when I call `netlist = AscEditor("asc/LPF_3rd.asc")`, I get the same error that voltage.asy is not found (I've verified that it is in the lib/sym folder.

NooriDan, 2025-02-04:
I tried running the branch in development mode (in a seperate virtual environment) but it gave me the same error. However, adding `"/Users/username/Documents/LTspice/lib"` I was able to read the .asc file however Im facing issue running simulations. I belive in the [tutorials](https://spicelib.readthedocs.io/en/latest/modules/run_simulations.html) they use SpiceEditor to read asc and convert to net file automatically but that fails giving the following error:

---------------------------------------------------------------------------
EncodingDetectError                       Traceback (most recent call last)
Cell In[22], line 1
----> 1 netlist = SpiceEditor("asc/LPF_3rd.asc")  # Open the Spice Model, and creates the .net
      2 netlist

File ~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/editor/spice_editor.py:1140, in SpiceEditor.__init__(self, netlist_file, encoding, create_blank)
   1138         self.encoding = detect_encoding(self.netlist_file, r'^\*')  # Normally the file will start with a '*'
   1139     except EncodingDetectError as err:
-> 1140         raise err
   1141 else:
   1142     self.encoding = encoding

File ~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/editor/spice_editor.py:1138, in SpiceEditor.__init__(self, netlist_file, encoding, create_blank)
   1136 if encoding == 'autodetect':
   1137     try:
-> 1138         self.encoding = detect_encoding(self.netlist_file, r'^\*')  # Normally the file will start with a '*'
   1139     except EncodingDetectError as err:
   1140         raise err

File ~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/utils/detect_encoding.py:80, in detect_encoding(file_path, expected_pattern, re_flags)
     78 else:
     79     if expected_pattern:
---> 80         raise EncodingDetectError(f"Expected pattern \"{expected_pattern}\" not found in file:{file_path}")
     81     else:
     82         raise EncodingDetectError(f"Unable to detect encoding on log file: {file_path}")

EncodingDetectError: Expected pattern "^\*" not found in file:asc/LPF_3rd.asc

NooriDan, 2025-02-04:
Follow up on the above: 

I converted the .asc to .net manually using LTspice GUI and tried calling SpiceEditor on the .net file. This part works well but when I try to run the simulation using LTspice runner I get the following error:

`Fatal Error: Could not open library file "LTC2.lib"`

I've set `runner.simulator._default_lib_paths` to 

`['~/AppData/Local/LTspice/lib',
 '~/Documents/LtspiceXVII/lib/',
 '~/Local Settings/Application Data/LTspice/lib',
 ['/Users/danialnoorizadeh/Documents/LTspice/lib']]`

Then I use `run_netlist_file = SpiceEditor("net/LPF_3rd.net")` and `runner.run_now(run_netlist_file)`  to run the simulation.

NooriDan, 2025-02-04:
I get this error on macos when I run 

`netfile_fname = LTspice.create_netlist("myfile.asc")  # in goes a .asc file, out goes a .net file`

---------------------------------------------------------------------------
NotImplementedError                       Traceback (most recent call last)
Cell In[3], [line 1](vscode-notebook-cell:?execution_count=3&line=1)
----> [1](vscode-notebook-cell:?execution_count=3&line=1) netfile_fname = LTspice.create_netlist("net/LPF_3rd.asc")  # in goes a .asc file, out goes a .net file

File ~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/simulators/ltspice_simulator.py:310, in LTspice.create_netlist(cls, circuit_file, cmd_line_switches, stdout, stderr)
    [306](https://file+.vscode-resource.vscode-cdn.net/Users/danialnoorizadeh/Code/Sandbox/SymXplorer/SpiceSizer/~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/simulators/ltspice_simulator.py:306) circuit_file = Path(circuit_file)
    [308](https://file+.vscode-resource.vscode-cdn.net/Users/danialnoorizadeh/Code/Sandbox/SymXplorer/SpiceSizer/~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/simulators/ltspice_simulator.py:308) if cls.using_macos_native_sim():
    [309](https://file+.vscode-resource.vscode-cdn.net/Users/danialnoorizadeh/Code/Sandbox/SymXplorer/SpiceSizer/~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/simulators/ltspice_simulator.py:309)     # native MacOS simulator
--> [310](https://file+.vscode-resource.vscode-cdn.net/Users/danialnoorizadeh/Code/Sandbox/SymXplorer/SpiceSizer/~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/simulators/ltspice_simulator.py:310)     raise NotImplementedError("MacOS native LTspice does not have netlist generation capabilities. Use LTspice under wine.")
    [312](https://file+.vscode-resource.vscode-cdn.net/Users/danialnoorizadeh/Code/Sandbox/SymXplorer/SpiceSizer/~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/simulators/ltspice_simulator.py:312) cmd_netlist = cls.spice_exe + ['-netlist'] + [circuit_file.as_posix()] + cmd_line_switches
    [313](https://file+.vscode-resource.vscode-cdn.net/Users/danialnoorizadeh/Code/Sandbox/SymXplorer/SpiceSizer/~/.pyenv/versions/myenv/lib/python3.12/site-packages/spicelib/simulators/ltspice_simulator.py:313) error = run_function(cmd_netlist, stdout=stdout, stderr=stderr)

NotImplementedError: MacOS native LTspice does not have netlist generation capabilities. Use LTspice under wine.

hb020, 2025-02-04:
oops indeed

hb020, 2025-02-04:
> Follow up on the above: .....

2 problems found. 

* the lib paths. I now set
```
    _default_lib_paths = ["~/AppData/Local/LTspice/lib",
                          "~/Documents/LTspiceXVII/lib/",
                          "~/Documents/LTspice/lib/",
                          "~/My Documents/LTspiceXVII/lib/",
                          "~/My Documents/LTspice/lib/",
                          "~/Local Settings/Application Data/LTspice/lib"]
```
That should work. You no longer should have to set `runner.simulator._default_lib_paths` (which you should have avoided anyway, you should have used `.set_custom_library_paths()`)

* the components (.asy) refer to `LTC2.LIB`,  but the file name is `LTC2.lib`. This MIGHT be a cause of problems.

Can you tell me the exact error message you get, and what tool gives that error? You no longer should have to set `runner.simulator._default_lib_paths` (which you should avoid anyway, use `.set_custom_library_paths()`)

NooriDan, 2025-02-05:
**Regarding this...**
> * the components (.asy) refer to `LTC2.LIB`,  but the file name is `LTC2.lib`. This MIGHT be a cause of problems.

Im not sure if capitalization is the problem because I installed wine and this has solved the problem. it's now picking up on wine only (not the MacOS version LTspice), reads the .asc schematic I shared earlier and successfully simulates it. 

I think the best course of action is to ask users to use wine instead of the macos version haha :)

**On a separate note...**
I was wondering how adding a new PDK would work for this tool (for transistor level simulations) works?

hb020, 2025-02-05:
see PR #138

Integration with a PDK would be a whole new step indeed.

hb020, 2025-02-05:
Branch is merged in main. Ticket will be closed when release is made.

NooriDan, 2025-02-06:
Thank you for your support! Keep up this project it’s very helpful :)

closed 2025-02-08 by nunobrum