nunobrum/spicelib #139: ReadRaw Fails to Parse Xyce-Generated .raw File on Ubuntu (Block Size Mismatch Error)

jiaxinmax, 2025-02-05:
Hello! When I use Xyce to run simulations and generate. raw files on Ubuntu, I encounter an error when trying to read the file using ReadRaw, which prevents me from correctly parsing the file content.

**Environmental information:**
Operating System: Ubuntu
Xyce version: 7.8
Reading tools: ReadRaw
. raw file source: Xyce simulation generated

**Steps:**
Use Xyce to run simulations on Ubuntu to generate. raw files.
Try using ReadRaw to read. raw files.
Reading failed with the following error:
Error in calculating block size. Expected 32 bytes, but found 60 bytes.

**Expected results:**
ReadRaw should successfully parse the. raw file generated by Xyce.

**Actual results:**
An error occurred and the. raw file cannot be parsed, even if it only contains one dataset.

We look forward to your reply. Thank you.

[error.zip](https://github.com/user-attachments/files/18667951/error.zip)

hb020, 2025-02-05:
Interesting file. It has a binary section, followed by a text section, and is a sensitivity analysis. 
Xyce is indeed not (yet) supported for raw file reading. 
Will need to do some digging to make this work properly. 
It will not be too complicated to read the binary part, but will need to read the source code of xyce to see what they wanted with the text part.

hb020, 2025-02-06:
The ASCII raw files should load now from branch "ascii-raw-....", but only the first/normal part will load. Will probably do the same for binary.

jiaxinmax, 2025-02-08:
Thank you very much for your response.

closed 2025-02-13 by nunobrum