nunobrum/spicelib #140: New LTspice version breaks various selftests

hb020, 2025-02-05:
at end of `test_spicelib.py` run:

```
======================================================================
ERROR: test_batch_test (__main__.test_spicelib.test_batch_test)
@note   inits class
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/workspace/spicelib/unittests/test_spicelib.py", line 103, in test_batch_test
    editor.set_component_value('XU1:C2', 20e-12)  # modifying a
    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^
  File "/workspace/spicelib/spicelib/editor/spice_editor.py", line 842, in set_component_value
    self._set_component_attribute(reference, 'value', value)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/workspace/spicelib/spicelib/editor/spice_editor.py", line 505, in _set_component_attribute
    sub_circuit_original = self.get_subcircuit(subckt_instance)  # If not will look for it.
  File "/workspace/spicelib/spicelib/editor/spice_editor.py", line 445, in get_subcircuit
    line_no = self.get_line_starting_with(subckt_ref)
  File "/workspace/spicelib/spicelib/editor/spice_editor.py", line 332, in get_line_starting_with
    raise ComponentNotFoundError(error_msg)
spicelib.editor.base_editor.ComponentNotFoundError: line starting with 'XU1' not found in netlist

======================================================================
ERROR: test_fourier_log_read (__main__.test_spicelib.test_fourier_log_read)
Fourier Analysis Test
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/workspace/spicelib/unittests/test_spicelib.py", line 524, in test_fourier_log_read
    log = LTSpiceLogReader(log_file)
  File "/workspace/spicelib/spicelib/log/ltsteps.py", line 338, in __init__
    dc_component = float(line.split(':')[-1])
ValueError: could not convert string to float: ''

======================================================================
FAIL: test_operating_point (__main__.test_spicelib.test_operating_point)
Operating Point Simulation Test
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/workspace/spicelib/unittests/test_spicelib.py", line 381, in test_operating_point
    self.assertListEqual(traces, [1.0, 0.5, 4.999999873689376e-05, 4.999999873689376e-05, -4.999999873689376e-05], "Lists are different")
    ~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AssertionError: Lists differ: [np.float64(0.5), np.float32(1.0), np.floa[45 chars]-05)] != [1.0, 0.5, 4.999999873689376e-05, 4.999999[33 chars]e-05]

First differing element 0:
np.float64(0.5)
1.0

+ [1.0, 0.5, 4.999999873689376e-05, 4.999999873689376e-05, -4.999999873689376e-05]
- [np.float64(0.5),
-  np.float32(1.0),
-  np.float32(5e-05),
-  np.float32(-5e-05),
-  np.float32(5e-05)] : Lists are different

----------------------------------------------------------------------
Ran 11 tests in 39.276s

FAILED (failures=1, errors=2)
```

hb020, 2025-02-05:
The "ComponentNotFoundError: line starting with 'XU1' not found in netlist" is because  now the netlist that gets created refers to "X§U1"

hb020, 2025-02-05:
See branch [ltspice_version_24_1](https://github.com/nunobrum/spicelib/tree/ltspice_version_24_1), WIP

hb020, 2025-02-05:
> I see this also in my files. I wonder if this is a bug on their end.

It is considered an optional character for other spice variants it seems. So I'll just scrap it for now. Is repaired in the new branch (not yet ready for merge, remaining other issues)

hb020, 2025-02-05:
Remaining problem: 
```\workspace\spicelib\unittests\output\Batch_Test.net_AD820_5.net(18): There is more than one sub-circuit definition with this name, which is ill-formed.```

This happens with the circuits that manipulate AD820 and produce `.subckt AD820_XU1`

nunobrum, 2025-02-05:
@hb020 
Ouch! This seems like a really big break in terms of compatibility. I can confirm that I was contacted directly by other users complaining that the new LTSpice breaks compatibility and brings a few problems. 
This weekend  I'll checkout the branch you have started and try the new LTSpice.
Unfortunately, I've just changed jobs and at this moment I don't have much time free during the week.

Thanks for all what you are doing.

Cheers,
Nuno

hb020, 2025-02-06:
See PR #143

hb020, 2025-02-06:
Branch is merged in main. Ticket will be closed when release is made.

closed 2025-02-08 by nunobrum