nunobrum/spicelib #152: Cannot upgrade to SpiceLib 1.3.8, remains on 1.3.6

womo75, 2025-02-09:
I tried to upgrade SpiceLib to version 1.3.8 using pip install --upgrade spicelib, but it did not update.

Then, I tried uninstalling and reinstalling the package, but the installed version remains 1.3.6.

Could there be an issue with the 1.3.8 release availability on PyPI?

nunobrum, 2025-02-09:
I think we can try to regress the python version back to 3.8.

nunobrum, 2025-02-09:
Unfortunately that is not possible. The method that we are using to create packages forced us to go to 3.12.
What you can do, if you can't upgrade to 3.12, is to use download the package to your computer and add it to your PYTHONPATH.

hb020, 2025-02-09:
maybe 3.9 will work?

hb020, 2025-02-10:
The reason poetry fails for 3.8 is "No matching distribution found for poetry==2.0.1". So the default install script does not work. But 3.9 works. @nunobrum if you don't have time for it, I can try to move the repo to 3.9.

nunobrum, 2025-02-10:
Good find. Will you try to create the release yourself?
I use these two commands:

`git tag x.y.z`
`git push origin --tags`

Good luck!

hb020, 2025-02-11:
Done:

python3.9 -m pip index versions spicelib
spicelib (1.4.0)
Available versions: 1.4.0, 1.3.6, 1.3.5, 1.3.4, 1.3.3, 1.3.2, 1.3.1, 1.3.0, 1.2.1, 1.2.0, 1.1.4, 1.1.3, 1.1.2, 1.1.1, 1.0.4, 1.0.3, 1.0.2, 1.0.1, 1.0.0, 0.9.1, 0.9, 0.8, 0.7, 0.6, 0.5, 0.4, 0.3, 0.2, 0.1

closed 2025-02-11 by hb020