nunobrum/spicelib #154: KeyError: '_SUBCKT' when processing schematic with Embedded Subckt

womo75, 2025-02-10:
I encountered an error when generating a netlist from a schematic (.qsch file) that contains an Embedded Subckt. During the netlist generation in the QschEditor, the following error is raised:

    436     pins = symbol_tag.get_items("pin")
    437     sub_ports = " ".join(pin.get_attr(QSCH_SYMBOL_PIN_NET) for pin in pins)
    438     subcircuits_to_write[model] = (
--> 439         component.attributes['_SUBCKT'],  # the subcircuit schematic is saved
    440         sub_ports,  # and also storing the port position now, so to save time later.
    441     )
    442 nets = " ".join(component.ports)
    443 netlist_file.write(f'{refdes} {nets} {model}{parameters}\n')

[test.zip](https://github.com/user-attachments/files/18738243/test.zip)

KeyError: '_SUBCKT'

hb020, 2025-02-20:
While testing, found that the type of subcircuit specification in your file is not supported yet:
```    
  �component (-4800,-1300) 0 0
    �symbol Proportional
      �type: X�
      �description: Proportional�
      �library file: |.subckt P OUT IN\nE1 OUT 0 IN 0 {P}\n.ends P�
      �shorted pins: false�
      �triangle (-100,200) (-100,-200) (300,0) 0 0 0x1000000 0x2000000 -1 -1�
      �text (-100,250) 1 6 2 0x1000000 -1 -1 "X2"�
      �text (0,0) 1 0 0 0x1000000 -1 -1 "P"�
      �text (0,-300) 1 0 0 0x1000000 -1 -1 "P=2"�
      �pin (300,0) (30,150) 0.63 13 0 0x0 -1 "OUT"�
      �pin (-100,0) (-100,10) 0.63 14 0 0x0 -1 "IN"�
    �
  �
```
Note the ` |.subckt P OUT IN\nE1 OUT 0 IN 0 {P}\n.ends P`

Qspice takes that (without looking for any libraries, therefore no need for 'Proportional.qsym') and transforms it into
```
.subckt X2•P OUT IN
E1 OUT 0 IN 0 {P}
.ends P
```
and
```
X2 OUT IN X2•P P=2
``` 

Spicelib for now only supports library files when confronted with `type: X`

Detaching this from branch `159-get_all_parameter_names-method` and making a new branch for it.

hb020, 2025-02-20:
NOTES: 
It is working for the provided test file, but I am convinced that we do not have complete coverage for all types of subcircuits at this stage. Not being a qspice expert, it would be best to have more .qsch and matching .net/.cir files to test upon.

@nunobrum or @womo75 do you have some more test files to share with subcircuits?

nunobrum, 2025-02-22:
@hb020 I have started to make a proper testbench for it. I have also made some changes.
I thought it would be better to solve the tag thing directly on the parsing code.

closed 2025-02-23 by nunobrum