nunobrum/spicelib #159: Suggestion for get_all_parameter_names Method

franzpl, 2025-02-19:
Hi,

I am missing a method that returns all parameter names from the netlist. Maybe this could be considered in the next version.

Inspired by the `_get_param_named()` method, I created a `get_all_parameter_names()` method, which looks as follows: 
```
   def get_all_parameter_names(self) -> List[str]:
        """
        Returns all parameter names from the netlist.

        :return: A list of parameter names found in the netlist
        :rtype: List[str]
        """
        param_names = []
        search_expression = re.compile(PARAM_REGEX(r"\w+"), re.IGNORECASE)
        for line in self.netlist:
            cmd = get_line_command(line)
            if cmd == '.PARAM':
                matches = search_expression.finditer(line)
                for match in matches:
                    param_name = match.group('name')
                    param_names.append(param_name)
        return param_names
```

I have tested this method, and it works as expected. Just an idea.

hb020, 2025-02-19:
Thanks. I'll add it, with versions for AscEditor, QschEditor and SpiceCircuit. (and unit tests)

hb020, 2025-02-19:
While working on this, stumbled upon #154 
Will add that in this branch