nunobrum/spicelib #163: Issue with Library Path in Windows

franzpl, 2025-02-21:
Hi,

I am facing an issue with the library path in my simulation. I am using Windows 11. 
Below is my current folder structure: 
```
test/
├── src/
│   ├── top_circuit.net
│   └── lib/
│       └── test_library.bib
└── test_script.py
```

I have adapted the `top_circuit.net` and the `test_library.lib` files from the `spicelib/examples/testfiles/top_circuit.net` file to illustrate my issue.

`top_circuit.net`
```
* C:

.inc test_library.lib

V1 N001 0 PULSE(0 1 1n 1n 1n 1u 1u 1)
R1 N002 N001 10
XX1 N002 N003 sub_circuit
R2 N003 0 1k

.tran 2u
.backanno
.end
```

`test_library.lib`
```
.subckt sub_circuit IN OUT
C1 OUT 0 1n
L1 IN OUT 1µ
.ends sub_circuit
```

When I run the `test_script.py` with the `output_folder` set to the `lib` folder, it works:
```
from spicelib import SpiceEditor
from spicelib import SimRunner
from spicelib.simulators.ltspice_simulator import LTspice

netlist = SpiceEditor(r'C:\test\src\top_circuit.net')
sim = SimRunner(simulator=LTspice, output_folder=r'C:\test\src\lib')
sim.run(netlist)
```

However, I want to output the simulation files in the `test` folder instead of the `lib` folder. 
According to the README, I should use the `set_custom_library_paths` method to set the library path.
So, I modified my code as follows: 
```
from spicelib import SpiceEditor
from spicelib import SimRunner
from spicelib.simulators.ltspice_simulator import LTspice

netlist = SpiceEditor(r'C:\test\src\top_circuit.net')
netlist.set_custom_library_paths(r'C:\test\src\lib')
sim = SimRunner(simulator=LTspice, output_folder=r'C:\test')
sim.run(netlist)
```
However, this generates a failed simulation with the error `Fatal Error: Could not open include file "test_library.lib"`

What am I missing here?

hb020, 2025-02-21:
Is that ltspice telling you `Fatal Error: Could not open include file "test_library.lib"`? In that case, `set_custom_library_paths` has no impact, because that is for spicelib. LTspice has its own search methods.
The cause of the message lies probably in the fact that the lib is inside a subdirectory. Try putting the lib in the same directory as the netlist, or use `.inc lib/test_library.lib` in the netlist.

franzpl, 2025-02-21:
Yes, it is from ltspice. I tested your approach before, but it had no effect. Neither placing the library in the same directory as the netlist, nor placing the library in the subdirectory using `.inc lib/test_library.lib` resolved the issue. The error message remained unchanged. The only solution that worked was setting the `output_folder` in SimRunner to the same path as the netlist. In that case, `.inc test_library.lib` in the same directory as the netlist and the library placed in the subdirectory `lib` using `.inc lib/test_library.bib` works.

nunobrum, 2025-02-23:
Hello @franzpl ,

I haven't tested this, but, you can try to add the 'lib' directory to the search paths dialog in LTSpice.

[image]

In alternative, I saw that the new LTspice version has a -I switch to add search paths, you can probably use it to solve your problem.
@hb020 is right, unfortunately the set_custom_library_paths() method is only working for the updates to the netlist.
It would make sense to automatically use it for setting the -I argument, but, for some reason, this need never came up before.

This is something to tackle in future versions.

franzpl, 2025-02-28:
Hi @nunobrum,

thank you for your response!

I tested your first approach using LTspice 24.1.4, and it worked. It is crucial that after adding the library search path and confirming with "OK", the GUI needs to be closed **once** for batch mode to successfully locate the library.

According to the -I switch, despite several attempts using the latest LTspice 24.1.4 version, I still can't get the simulation to run.  First, I tried the following command

```
.\LTspice.exe -b -run -IC:\test\src\lib "C:\test\src\top_circuit.net"
```

but still getting the `Fatal Error: Could not open include file "test_library.lib"` from LTspice. 

I opened an issue in the LTspice forum and was advised that the -I switch must be the last specified parameter. 

https://ez.analog.com/design-tools-and-calculators/ltspice/f/q-a/592920/batch-mode-simulation-with-external-library-path-issue

This is also noted in the LTspice24 manual:

_-I<path> | Specify a path to insert in the symbol and file search  paths. Must be the last specified option. No space between "-I" and  <path> is allowed._

Despite this, the following command: 

```
.\LTspice.exe -b "C:\test\src\top_circuit.net" -IC:\test\src\lib
```
is still not working for me. My issue in the forum remains unresolved .. 

While I'm fine with the GUI approach, I'm interested in getting the -I switch to function correctly. So, if you have any other ideas, please let me know. 

Thank you.

hb020, 2025-02-28:
Since you're on windows, 

`.\LTspice.exe -b "C:\test\src\top_circuit.net" -IC:\test\src\lib`

should probably be 

`.\LTspice.exe -Run -b "C:\test\src\top_circuit.net" -IC:\test\src\lib`

nunobrum, 2025-03-02:
Maybe another workaround would be to configure a LTspice.ini and give it as parameter. 
This has the advantage of allowing you to set much more than the library search paths.

franzpl, 2025-03-07:
Hello everyone,

@hb020 : Your suggested approach:
```
.\LTspice.exe -Run -b "C:\test\src\top_circuit.net" -IC:\test\src\lib
```
did not work either. In the meantime, I received an update in the LTspice forum telling me that it does not work due to a regression bug that will be fixed in the next version (> 24.1.4).

@nunobrum: That's a really good idea. I hadn't considered that yet.

Thank you both for your support. As my issue is answered now, I am going to close this issue.

closed 2025-03-07 by franzpl