nunobrum/spicelib #164: Regex does not cover ".5" as valid number

bam1rt, 2025-02-21:
https://github.com/nunobrum/spicelib/blob/4015a4af714a7a8b98cfcb35824c65babb4c1a75/spicelib/editor/base_editor.py#L71

I use ".5" to specify "0.5" or "500m". Adding a point to the regex solved it for me.

`    return r"(?P<name>" + pname + r")\s*[= ]\s*(?P<value>(?P<cb>\{)?(?(cb)[^\}]*\}|[\d\+\-Ee]+[a-zA-Z%]*))"
`
`    return r"(?P<name>" + pname + r")\s*[= ]\s*(?P<value>(?P<cb>\{)?(?(cb)[^\}]*\}|[\d\+\-Ee**.**]+[a-zA-Z%]*))"
`

nunobrum, 2025-02-22:
I think you should escape the . because it is a reserved character in regular expressions that mans any character.

closed 2025-02-22 by nunobrum

bam1rt, 2025-02-22:
I am using PyLTSpice 5.4.0 build Oct. 24
[image]

hb020, 2025-02-22:
what version of spicelib do you use? pyltspice is just a front-end

bam1rt, 2025-02-22:
1.3.5. OK, will try to update to 1.4. Thanks for your support !