nunobrum/spicelib #165: How to specify solver and path to symbols?

JeffKayzerman, 2025-02-26:
I am trying to run a simple simulation that depends on symbols stored in a share drive. Some simulations we want to use the normal solver, others we want to use the alternate solver. How can we make these changes per run?

hb020, 2025-02-26:
About the symbols: 
* if you want to make them available to any of the editors, you need to use `set_custom_library_paths() ` for example.
* if you want to make them available to the simulators, you need to use command line parameters that you can pass via the run command (see more below)

About the normal solver and alternative, I gather that you want to use ltspice. Quote from ltwiki: "There is no .option to specify which solver is used, the choice must be made before the netlist is parsed because the two solvers use different parsers.". To specify the solver, you therefore must pass that as command line parameters (`-alt` or -normal` in this case) to the run command. See https://spicelib.readthedocs.io/en/latest/classes/ltspice_simulator.html or https://spicelib.readthedocs.io/en/latest/classes/sim_runner.html, depending on the method you use to run the simulator. Both pages mention the 'alt' solver.

JeffKayzerman, 2025-02-26:
Thank you @hb020 I am aware of the command line arguments but using PyLTspice we are not using the command line arguments we are using the SimRunner class in python. While the documentation does give some examples on how to use the classes the class definitions themselves have blank documentation

hb020, 2025-02-26:
PyLTspice has become not much more than a wrapper around spicelib, and indeed, documentation lacking in places. Even if you use ltspice exclusively, you will find that you can easily move over to spicelib. It also defaults to ltspice in many cases. Can you try spicelib directly?

JeffKayzerman, 2025-02-27:
Reading up on spicelib now. Let me back up and explain what I am trying to do. The goal of my application is to offload LTspice workloads to a server so the user does not have to run them on their local machine. 

**System:**
I have a web application that lets the user drag and drop a simulation file from their local machine. The web application hands off the file to a python process running on a server. The python process saves the file locally and kicks off LTspice via the command line interface. When LTspice finishes, the python process places the output files in a share drive that the user can access and they are emailed that the simulation is complete. This process is asynchronous so many simulations can be run at once and once the server runs out of resources the rest of the simulations sit in a queue until resources are freed up. 

All of this is working great when I am logged into the server and running it under my account. Now when I try to run the python process as a windows service under a service account it reports an error when trying to run LTspice. The error code is 3221226505. I've tried to check that the service account has all the permissions to every folder it needs to access but I still get this error. It seems to be something when LTspice tries to load symbols because if I leave the symbol search paths blank it comes back with a "Missing these symbols" message instead of the error code.

So... this has led me to try to take a different approach and try PyLTSpice until now when I just learned about spicelib. Will see if I can accomplish the same thing. If you have any other advice it is greatly appreciated.

hb020, 2025-02-27:
For the vast majority of things, where you user 
```python
import PyLTSpice
```
you can now put
```python
import spicelib
```
And the code will continue working.

About the web site approach, are you aware of the client/server functionality in spicelib? See https://github.com/nunobrum/spicelib?tab=readme-ov-file#run_serverexe for an example on how to use a spicelib server.

Or you can look in the client/server classes to make your own.

JeffKayzerman, 2025-02-27:
Thank you for your help, this is my first time using spicelib so please bear with me. I tried to just write the simplest script (shown below. *Note* my LTspice installation is 24.0.12

`runner = SimRunner(simulator=LTspice, output_folder='./temp')
netlist = SpiceEditor(fpath)
        opts = []
        if solver == "alternate":
            opts.append('-alt')
        else:
            opts.append('-norm')

        runner.run(netlist, switches=opts, exe_log=True)`

I opened my simulation in LTspice and exported it as a netfile. These are the first few lines of that file

`"ExpressPCB Netlist"
"LTspice"
1
0
0
""
""
""
"Part IDs Table"`

When I try to run it I get this error

`Task exception was never retrieved
future: <Task finished name='Task-2' coro=<run_ltspice_batch() done, defined at C:\Temp\LTSpice-server\spicelib-server.py:114> exception=EncodingDetectError('Expected pattern "^\\*" not found in file:NED_Stability_Flyback_Basso_27V.net')>
Traceback (most recent call last):
  File "C:\Temp\LTSpice-server\spicelib-server.py", line 123, in run_ltspice_batch
    netlist = SpiceEditor(fpath)
  File "C:\Temp\LTSpice-server\.venv\Lib\site-packages\spicelib\editor\spice_editor.py", line 1157, in __init__
    raise err
  File "C:\Temp\LTSpice-server\.venv\Lib\site-packages\spicelib\editor\spice_editor.py", line 1155, in __init__
    self.encoding = detect_encoding(self.netlist_file, r'^\*')  # Normally the file will start with a '*'
                    ~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Temp\LTSpice-server\.venv\Lib\site-packages\spicelib\utils\detect_encoding.py", line 80, in detect_encoding
    raise EncodingDetectError(f"Expected pattern \"{expected_pattern}\" not found in file:{file_path}")
spicelib.utils.detect_encoding.EncodingDetectError: Expected pattern "^\*" not found in mySimulation.net`

When I try to put a comment at the top of the netlist file I get this error instead

`Task exception was never retrieved
future: <Task finished name='Task-3' coro=<run_ltspice_batch() done, defined at C:\Temp\LTSpice-server\spicelib-server.py:114> exception=SyntaxError('Unrecognized command in line: ""ExpressPCB Netlist"\n"')>
Traceback (most recent call last):
  File "C:\Temp\LTSpice-server\spicelib-server.py", line 123, in run_ltspice_batch
    netlist = SpiceEditor(fpath)
  File "C:\Temp\LTSpice-server\.venv\Lib\site-packages\spicelib\editor\spice_editor.py", line 1160, in __init__
    self.reset_netlist(create_blank)
    ~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^
  File "C:\Temp\LTSpice-server\.venv\Lib\site-packages\spicelib\editor\spice_editor.py", line 1274, in reset_netlist
    finished = self._add_lines(lines)
  File "C:\Temp\LTSpice-server\.venv\Lib\site-packages\spicelib\editor\spice_editor.py", line 338, in _add_lines
    cmd = get_line_command(line)
  File "C:\Temp\LTSpice-server\.venv\Lib\site-packages\spicelib\editor\spice_editor.py", line 155, in get_line_command
    raise SyntaxError(f"Unrecognized command in line: \"{line}\"")
SyntaxError: Unrecognized command in line: ""ExpressPCB Netlist"
"`

hb020, 2025-02-27:
Starting to see a problem here. I may be wrong, but I think you are misled into thinking that an expresspcb netlist and a ltspice netlist are just netlists and can be the same file. 

That is not the case. The first tool is a CAD tool and creates netlists that are mainly there to list connections between pins of devices. The latter is a pure simulation tool that has an extremely basic circuit design tool as a cherry on top. They have different requirements, and different formats.

To simplify things, if you have a schema and you want to create a PCB from it, you need to specify the packages of all the components, create a netlist, and send that netlist to the PCB CAD tool.

If you have a schema and want to simulate it, you need to specify simulation models (spice models) for all of the components, create a netlist, and send it to the simulator.

As you can see, both paths (schema editor -> PCB design tool, and schema editor -> simulator) use files called netlists. But the formats and requirements are different.

Not many PCB tools can provide both types of netlists. In the more 'affordable' rangem there is Kicad (it uses ngspice, not ltspice). And it does not provide 1 file that can go to both PCB and to simulation, it keeps the 2 types separate.

Now the good news is that simulator netlists have a relatively standard format, and simple netlists can be read and simulated in wide range of simulator tools. spicelib takes advantage of that, and can interface with 4 different simulators.

I don't know expresspcb, and maybe it can export to ltspice netlists, but to give you an idea of what ltspice netlist files look like:

Simple RC filter:
```
* C:\sandbox\spicelib_dev\examples\testfiles\TRAN.asc
Vin in 0 PWL(0 0 10n 1)
R1 in out 1k
C1 out 0 1u
.tran 5m
.backanno
.end
```
(ltspice knows the models of R and C, so no need to specify the models. For opamps etc it is another thing)

That does not look like what you mentioned, as far as I understand, which was 
```
ExpressPCB Netlist
LTspice 1 0 0 

Part IDs Table
```

(block quotes here = surround by 3 times backtick)

JeffKayzerman, 2025-02-27:
Hmm. In LTspice I open my .asc file, then go to tools "export netlist". By default it wants to save it as ExpressPCB (*.net). When I click the dropdown for save as type there are many other options that are all still .net files. Allegro, Callay, Computervision, EE Designer, Multiwire, Vectron, Wire List are just a few of the options. Which one should I be choosing? Unfortunately I can't share the actual simulation file as it is proprietary

JeffKayzerman, 2025-02-28:
Thank you that worked while running on my server under my account. Unfortunately when I try to run it as a windows scheduled task or windows service the python runs but it seems ltspice does not execute. I'm beginning to think its not possible to run ltspice this way.

hb020, 2025-02-28:
Well, that should have been simply
```run_server --port 9000 --parallel 4 --output ./temp LTSpice 300```

but there is a small bug (on macos, and likely also on windows), but you can also run 

```python3 -m spicelib.scripts.run_server  --port 9000 --parallel 4 --output ./temp LTSpice 300```

JeffKayzerman, 2025-02-28:
The line I used came right out of the documentation but the one you provided actually worked so thank you. I am going to play around with this approach to see if I can get it to work well. I have one concern though.

My engineers tell me that they have other files (.iso, .txt) that need to be in the same folder as the schematic in order for the simulation to work. Will that work with this server based approach?

nunobrum, 2025-03-02:
I have merged the PR #168 . 
Will close this as soon as it makes part of a release.

JeffKayzerman, 2025-03-03:
@hb020 thank you for all your help. I finally got it working using the LTspice class. For some reason that class seems to be able to execute LTspice while python is running as a service whereas running it directly it would fail. I am all set now and very happy to have learned about this library, will be a big help in our future work. Thanks!

closed 2025-03-03 by hb020