nunobrum/spicelib #166: Integrating run_filename parameter in SimStepper

franzpl, 2025-02-28:
Hi, 

the `run` method in SimRunner allows overriding the netlist name using the `run_filename` parameter:

```
def run(self, netlist: Union[str, Path, BaseEditor], *, wait_resource: bool = True,
            callback: Union[Type[ProcessCallback], Callable] = None,
            callback_args: Union[tuple, dict] = None,
            switches=None,
            timeout: float = None,
            run_filename: str = None,
            exe_log: bool = False) -> Union[RunTask, None]:
```

Currently, this `run_filename`  is missing in the `run_all` method of SimStepper:

```
def run_all(self,
                callback: Union[Type[ProcessCallback], Callable] = None,
                callback_args: Union[tuple, dict] = None,
                switches = None,
                timeout: float = None,
                use_loadbias='Auto',
                wait_completion=True) -> None:
```

To address this, I added the `run_filename` parameter to the `run_all` method and updated the call to `self.runner.run`

```
self.runner.run(self.netlist, callback=callback, callback_args=callback_args,
                            switches=switches, timeout=timeout, run_filename=run_filename) 
```

However, I got an issue when using the `run_all` method where the simulation steps were overriden due to the same netlist name being used.  So, I modified the `_prepare_sim` function to include and `else` statement, so that the filename updates with `self.runno`

```
def _prepare_sim(self, netlist: Union[str, Path, BaseEditor], run_filename: str):
        """Internal function"""
        # update number of simulation
        self.runno += 1  # Incrementing internal simulation number
        # Harmonize the netlist into a Path object pointing to a netlist file on the right output folder
        if isinstance(netlist, BaseEditor):
            if run_filename is None:
                run_filename = self._run_file_name(netlist.circuit_file)
            else:
                run_filename = self._run_file_name(run_filename)
```
It works, but there's a drawback: when the `run` method is called with the `run_filename` parameter (e.g., 'test.net'), the netlist name receives a `_1` (e.g., `test_1.net`). 

As an alternative approach, the `_prepare_sim` function needs to be informed when SimStepper is activated.

hb020, 2025-02-28:
> To address this, I added the run_filename parameter to the run_all method and updated the call to self.runner.run

Sounds OK to me.

> So, I modified the _prepare_sim function to include and else statement, so that the filename updates with self.runno
>As an alternative approach, the _prepare_sim function needs to be informed when SimStepper is activated.

let me rephrase that last: if SimStepper is used with run_filename, make sure run_filename becomes unique when given to the simulator.

Can be interesting to be added. @nunobrum your idea?

nunobrum, 2025-03-02:
The run_filename was thought to be used the the run_itself. 
If you really need to have a filename override on the stepper, I would just add it to the stepper.run_all() method as you did, and make there all the processing so it is passed as an argument to the SimRunner.run().
My question is why do you need this ? If I understand your need I can better implement something that makes sense.

franzpl, 2025-03-07:
Hi @nunobrum,

if I could override the filename in SimStepper, I would be able to achieve better organisation when running multiple simulations with different parameters, so the output would reflect the particular run. This would allow me to plot the changed parameters more specifically.

nunobrum, 2025-03-08:
Hi @franzpl,

Now I understand what you mean. But, I'm afraid that providing the run_filename will not get the results you want. 
The call to the run_all makes the class Stepper to iterate on all the steps that were programmed, and in order to have that information on the name, the stepper needs to have access to the value that are in each iterator. 
This implementation needs to be done on the Stepper class itself.  More on that later.

Meanwhile, there are a couple of ways to workaround that.

1. The easiest way is not to use the Stepper class and use loops on your code, but duh... kind of defeats the purpose of the Stepper class
2. Use .MEAS to save to your log file the parameters that are being changed
3. Pass the netlist back to the callback in order to get back the parameters you changed and rename the files

I don't like any of these workarounds. None of that is elegant. My proposal to solve this, is to pass a callable function to run_all() that receives a dictionary as argument and that returns a filename in a string format. 
So you can pass something like this:
`stepper.run_all(run_filename ="run_{V1}_{R1}_".format)`

Another option, is that Stepper will have an attribute with a list of dictionaries with all this information. Probably also an option to dump tis information into a .csv file.
What do you think would be more convenient?

Best Regards,
Nuno

franzpl, 2025-03-08:
Hi @nunobrum,

now you have thought further than I had assumed. For my purposes, it's absolutely sufficient if I could use a simpler approach like this:
```
steps = {
          'V1': (3,4,5),
          'freq': (100, 1000, 10000)
}

Stepper = SimStepper(SpiceEditor("test.net"), runner)

for param_name, param_value in steps.items():
          Stepper.add_value_sweep(param_name, param_value)
          Stepper.run_all(run_filename=f'run_{param_name}')
```
Now, I would be able to organize the simulation runs in the same directory and to write a plotter function that only plots the traces of the RAW files where the voltage ‘V1’ or the frequency ‘freq’ has changed. So, for me, there is no need for SimStepper to have access to the values for each iteration. But I see the point, that I could solve this also without SimStepper by using loops and updating the `run_filename` in the `run()` method from SimRunner. What do you think? 

Best regards,
Franz

nunobrum, 2025-03-08:
Hi @franzpl ,

The idea of the SimStepper was to make combinations of all possible steps. Using your example. If you do this:
```python
steps = {
          'V1': (3,4,5),
          'freq': (100, 1000, 10000)
}

Stepper = SimStepper(SpiceEditor("test.net"), runner)

for param_name, param_value in steps.items():
      Stepper.add_value_sweep(param_name, param_value)
Stepper.run_all()
````
It would do all combinations of:
(V1 , freq) in (3, 100), (3, 1000), (3, 10000), (4, 100), (4, 1000), (4, 10000), (5, 100), (5, 1000), (5, 10000)

Also, I should tell you that it creates a simulation file for each combination. This is why just giving the run_filename as you suggest wouldn't work. You need a different filename for these 9 combinations.

To do what you want to do, which is a simple 1 dimensional sweep, you don't need the SimStepper. You can simply do this:

```python
steps = {
          'V1': (3,4,5),
          'freq': (100, 1000, 10000)
}

netlist = SpiceEditor("test.net")
runner = SimRunner(netlist)

for param_name, param_values in steps.items():
    for param_value in param_values:
        netlist[param_name]=param_value
        runner.run(netlist, run_filename=f'run_{param_name}_{param_value}')

# then this
runner.wait_completion()
# or this:
for raw_file, log_file in runner:
    # make your processing here
    rawdata = RawRead(raw_file)
````
This only makes 3 + 3 simulations and files are
run_V1_3, run_V1_4, run_V1_5, run_freq_100, run_freq_1000, run_freq_10000

Anyway, I see that for post processing, having the association between files and parameter is useful. I'll implement the mentioned improvement on the SimStepper. Although, I should say that the file naming won't be a practical implementation, as it obliges the user to make sure that the file_namer function cover all sweeps, and also, there are probably values that won't generate OS compatible filenames.

I'll close this issue as soon as the implementation "sneaks" into the main branch.

Best Regards,
Nuno

nunobrum, 2025-03-08:
Integrated on 1.4.2

closed 2025-03-08 by nunobrum