nunobrum/spicelib #175: TypeError when instantiating SimStepper

franzpl, 2025-03-09:
Hi @nunobrum,

I'm testing the new version, but I encounter a TypeError when instantiating the SimStepper class. I executed the 
 `sim_stepper_example.py`. The error message is: 

`
Stepper = SimStepper(SpiceEditor("./testfiles/Batch_Test.net"), runner)
`
`
TypeError: Can't instantiate abstract class SimStepper without an implementation for abstract methods 'add_instruction', 'remove_Xinstruction', 'remove_instruction', 'set_component_value', 'set_element_model'` 

Best regards,
Franz

nunobrum, 2025-03-09:
Oh! Jezzz. I'll fix that ASAP.

franzpl, 2025-03-10:
Hello again @nunobrum,

here are the steps I've followed:

Uninstalled spicelib: 

`pip uninstall spicelib`

 Result: received `ModuleNotFoundError: No module named 'spicelib'` when running the script.

Reinstalled spicelib: 

`pip install spicelib`

Result: `Successfully installed spicelib-1.4.2`, also confirmed with `pip show spicelib`:

`Version: 1.4.2`

However, I still get the same TypeError when instantiating SimStepper when running the script. 

I compared the locally installed sim_stepping.py with the version from the GitHub repository. They are identical.

I am using Python 3.13.2

Any other ideas on what I could test? 

Best regards,
Franz

nunobrum, 2025-03-10:
Are you sure that:
- you installed it on the same Python installation you are using
- you don’t have another spicelib checked out in your PYTHONPATH
- if you’re using GitHub, make a reset to the whole repo.

just in case, you can delete the AnyRunner protocol from the SimStepper class declaration.
That will make that error go away.

franzpl, 2025-03-17:
Hi @nunobrum,

yes, I verified your points, and I'm using the latest version. Regarding your suggestion, I indeed found that removing AnyRunner from 

`class SimStepper(AnyRunner):`

solves the issue.

I also tested the version 1.4.1, where the class declaration is:

`class SimStepper(object):`

In this version, I encountered no problems. But using AnyRunner in the class declaration leads to the same issue. 

Additionally, in the latest version, I found that removing the @wraps() decorators from the methods in the SimStepper class also solves the issues.

hb020, 2025-03-19:
he is using 3.13.2.

```
# python3 --version
Python 3.13.2
# python3 sim_stepper_example.py 
Traceback (most recent call last):
  File "/Users/me/workspace/spicelib/examples/sim_stepper_example.py", line 15, in <module>
    Stepper = SimStepper(SpiceEditor("./testfiles/Batch_Test.net"), runner)
TypeError: Can't instantiate abstract class SimStepper without an implementation for abstract methods 'add_instruction', 'remove_Xinstruction', 'remove_instruction', 'set_component_value', 'set_element_model'
```

hb020, 2025-03-19:
See https://github.com/python/cpython/issues/87176

One of the solutions would be to add `, updated=()` to all the `@wraps` decorators that are mentioned. Looks like that works.

nunobrum, 2025-03-19:
Nice catch. The interesting thing for me is that neither me nor at least the release unittests have the same problem.
I think we can live without the abstract classes. It is just a formalism that doesn't have any functional benefit.

hb020, 2025-03-21:
@nunobrum do you want me to pick this up?
I'd still prefer to keep the `class SimStepper(AnyRunner)` but either remove the `@wraps`, or add `, updated=()`

nunobrum, 2025-03-22:
@hb020, I trust your judgement. I have used the `updated=()` . 
Can you please review this [Pull request 190](https://github.com/nunobrum/spicelib/pull/190) ?

Note: It includes a shameless promotion of another tool I made. 
Hesitated for a long time to include it, but, I think it will be useful for the community. Let me know if you are OK with it.

nunobrum, 2025-03-22:
Fixed in 1.4.3

closed 2025-03-22 by nunobrum