nunobrum/spicelib #177: Error when retrieving resistor current with rawfile.get_trace

womo75, 2025-03-14:
When attempting to retrieve the current of a resistor using rawfile.get_trace, the following error occurs:

NotImplementedError: Unrecognized alias type for alias: "Time"

hb020, 2025-03-14:
Do you have an example of a raw file with that problem, and a code example? 
Are you sure the trace name you want to read is also available via `get_trace_names()`? 
The time axis is implicit.

womo75, 2025-03-14:
Here is an example circuit.

[test.zip](https://github.com/user-attachments/files/19251739/test.zip)

hb020, 2025-03-14:
Confirmed problem. This should work, at least when we look at functional equivalence with ltspice raw files. Looking into it.

hb020, 2025-03-14:
The cause was in the fact that LTspice and Qspice behave differently here. LTspice made `I(R1)` into a trace, and qspice made it into an alias. The alias handling was a bit off. See PR #180

closed 2025-03-14 by nunobrum