nunobrum/spicelib #179: ltspice rawread for binary files seems to mess up

hb020, 2025-03-14:
images speak louder than words:

[image]
[image]

```python
fnames = [
    "tran_ltspice.ascii.raw",
    "tran_ltspice.bin.raw",
]
for fname in fnames:
    print(fname)
    rawfile = RawRead("../../examples/testfiles/" + fname)
    print(rawfile.get_trace_names())
    # I_R1 = rawfile.get_trace("I(R1)")
    tr = rawfile.get_trace("V(OUT)")
    time = rawfile.get_trace("time")
    plt.clf()
    plt.plot(time, tr)
    plt.xlabel("Time (s)")
    plt.ylabel("Voltage (V)")
    plt.title(fname)
    plt.savefig(fname + ".png")
```

nunobrum, 2025-03-14:
This is known for a very long time. The problem is coming from LTSpice itself. 

This is why you should use get_timr_axis which takes the abs of the time

hb020, 2025-03-14:
Indeed, better `get_axis()`, since `get_time_axis` is deprecated. But anyway, yes. Closing.