nunobrum/spicelib #184: qspice editor remove_instruction and remove_Xinstruction also remove comments

hb020, 2025-03-16:
Minor issue, but best to be complete.

nunobrum, 2025-03-16:
I'll have a look into it.

hb020, 2025-03-16:
was working on it, but if you want to take it, go ahead.

nunobrum, 2025-03-16:
Oh! Go ahead.

nunobrum, 2025-03-16:
I was only starting. Still looking at the code. There the skip comment is used in a few spots. I was considering what would be the most elegant way to do it.

hb020, 2025-03-16:
I've done it the same way as qspice for now. Looking at parameters now, as they might have the same problem. Then I'll add unit tests.

hb020, 2025-03-16:
@nunobrum Strange thing I found: `remove_instruction()` states: 

```
This only works if the instruction exactly matches the line on the netlist. This means that space characters,
and upper case and lower case differences will not match the line.
```

But both qspice editor and asc_editor use `if instruction in text:` constructions, where I would expect `if instruction == text:`

Should this be corrected?

nunobrum, 2025-03-16:
I think I used that construct because I was afraid that comments could exist after the command. 
It's going to be a delicate balance between having something that is bullet proof and something simple.~
A bullet proof would require a complete syntax parser, and this is something I have been avoiding all along.
On the other hand, in most cases, the person who is removing the instruction is the person who added it in the first place.
So, kind I guess it is kind of OK if we relax a bit.
What do you think ?

hb020, 2025-03-16:
yes, it risks to break some existing users. Or one could add a "strictness" parameter that is "loose" by default.

Other small issue I found: the approaches for the `remove_Xinstruction` problem of "how to delete items from a list while I'm iterating through it" are very different between the qsch and asc editors. It works, but it took me some time to compare them. Maybe best to standardise on 1 method?

Not necessarily now. 

But so far I think the correct handling of comments is OK now for qsch and asc. Will look at the other editors, in how far they risk being affected, and them will add a unit test.

nunobrum, 2025-03-16:
I'll have to take a deeper look into it to understand what you are saying. I'll have a go at it another day.

closed 2025-03-17 by nunobrum