nunobrum/spicelib #192: Handle timeout exceptions in RunTask

arnaudgabas, 2025-03-28:
Hi,

I was looking for a way to catch/retrieve the `subprocess.TimeoutExpired` exception once a simulation reaches the specified timeout. Maybe it's a quick fix:

```
try:
    self.retcode = self.simulator.run(self.netlist_file.absolute().as_posix(), self.switches, self.timeout,
                                    exe_log=self.exe_log)
except subprocess.TimeoutExpired as e:
    self.print_info(_logger.error, e)
    self.retcode = -1
self.stop_time = clock_function()
```

Is there a specific way of retrieving exceptions generated by `RunTask` once using `SimRunner` methods (`run`, `run_now`) ?

hb020, 2025-03-29:
Yes, outside of the direct simulatorclass.run() calls there isn't much info on why/if there was an exception.

See the linked branch. 

@nunobrum should we set up a proper feedback of the return value and all kinds of exceptions? This is bigger than the timeout.

In the meantime, the linked branch captures the exceptions.

nunobrum, 2025-03-30:
I agree that this approach is minimalist. But, the question is where to put the cursor between nothing and everything.
I think that either case, it is up to the user to diagnostic the problem and solve it. We just should give just enough for the user to do it. 

A few ideas.

What about adding an error code attribute to the RunTask ?
The current approach was to rename the .log into .fail in case of error. Do we keep this ? At the time of implementation, it was helpful to quickly verify how many of the simulations did fail.
If keeping this, does make sense rename it to .timeout? 

In case there are 3 netlists that failed in 200, it could be interesting give an option to the user to try to repeat only the failing ones. This would be cool, but risks being a complete overkill and making it too complex.

What about creating a detailed logfile ?

nunobrum, 2025-04-02:
Hi @hb020 ,

I would prefer the idea of the log file.
There are two reasons that make me back from the callback implementation:
   1. Most people don't even use it. It's way more advanced than what most people expect and really need.
   2. The reason I implemented callbacks, was an optimization. An extension of the run process was to be able to execute also that processing in parallel. I really don't see it for error handling.

So, I would definitely add something to the log file and in addition to it. 
  a) Include on RunTask class a status flag and a function that returns the error information.
  b)  The SimRunner class already has a list of completed tasks and a counter for the number of failed tasks. The logical would be to also keep track of the failed tasks.

What do you think?

hb020, 2025-04-03:
The log file is complicated. Would you add info to the simulator's log file, or to the exe_log. The latter would be best for timeouts I think, as the simulator's log file has a parser and a structure that I'd prefer not to mess with. The other one is free text anyway.

Using failSim and enhancing RunTask is a good idea indeed. Why not keep runtask retcode as -2 (or something not positive, nor the default -1) for an exception, and add a string property with "exception cause": the name of the exception. Like 'subprocess.TimeoutExpired' or 'Simulator executable not found.' A string that might be parsed is best.

By the way, there were a couple of bugs: 
* if all tasks fail, `wait_results()` hangs if return code = -1. So I changed the exceptions to -2 in case of an exception.
* if you set `runner.run(....timeout=0.001)` you still got the exception `spicelib.sim.sim_runner.SimRunnerTimeoutError: Exceeded 600.0 seconds waiting for tasks to finish`. 

Corrected them in the branch.

nunobrum, 2025-04-04:
I like your retcode suggestion. It is simple and practical.

nunobrum, 2025-04-04:
@hb020 How do you want to do this?
Are you motivated to do it?

hb020, 2025-04-04:
yes, can do. Is easy. Bit busy coming days, but by end next week it will be done.

hb020, 2025-04-05:
The runner iterator, `get_result()`, `wait_results()` and `run_now()` all return `Tuple(raw_file, log_file)` when success. They deviate in reply or behaviour (like raise exceptions) when the simulation failed or returned something else than 0.

I see no direct reading of the retcode in our examples.

Ideally, we would also get the retcode and the error string out via the above methods.

Modifying the iterator is going to be hard if we want to keep it kind of retro compatible.
I suggest adding the retcode and error string to new versions of the 3 functions above: `get_result(full_result=True)`, `wait_results(full_result=True)` and `run_now(full_result=True)`

OK?

nunobrum, 2025-04-06:
In this case, it may be worth to break compatibility.
Or just have a different function name than to add a new parameter.

hb020, 2025-04-06:
yes, will go for the additional function.
(and by the way, reply via email tends to make you miss out on the latest edits on the comments, which I tend to do a lot)

nunobrum, 2025-04-06:
Yes, I see what you mean. And the post comes with a lot of trash.
I normally see the notification on the phone and use the computer to reply directly on GitHub.
This time was on a hurry and decided to just reply on the phone because I was on the go.
Next time I'll use the GitHub App on the phone, instead of using the reply feature.

closed 2025-04-07 by nunobrum