nunobrum/spicelib #194: Calculation Speed Issues while Using PyLTSpice for LTSpice Automation

DavitKharshiladzeTrain, 2025-03-29:
Hello,

I am using  LTSpice Python libraries and I had issues with the calculation speeds. It always takes approximately 1 second to complete the simulation, regardless of circuit complexity.

I have constructed a simple circuit and a snippet to portray the problems I am having, and I am attaching this simple code below.

Is there any way to improve this kind of behaviour?

import time
from PyLTSpice import SimRunner
from spicelib.simulators.ltspice_simulator import LTspice

# Create the netlist file (simple_lc.cir)
netlist_content = """* Simple LC Resonant Circuit
V1 N001 0 DC 1 AC 1
L1 N001 N002 10uH
C1 N002 0 1pF
R1 N002 0 50
.ac lin 1500 40Meg 60g
.save V(N002)
.end
"""

# Save netlist
netlist_filename = "simple_lc.cir"
with open(netlist_filename, "w") as f:
    f.write(netlist_content)

LTC = SimRunner(output_folder='/temp', simulator=LTspice) # init simulation runner

for i in range(1,10): # iterate ove reverything
        t1 = time.time()
        LTC.run(netlist=netlist_filename)
        LTC.wait_completion()
        print(f"{t1 - time.time()} - Simrun")

gudnimg, 2025-03-29:
It's likely this line https://github.com/nunobrum/spicelib/blob/ada1904c96eb6dac26dbef000c1eca3025c01405/spicelib/sim/sim_runner.py#L574

Even when I remove it, the 1 second runtime only drops to 0.5 seconds. It looks like to me the simulation takes around ~0.5 seconds.

Note however that removing the sleep function or even just lowering the sleep time will make the code be run more often and so increases CPU load.

hb020, 2025-03-29:
There are 2 ways to lower the minimal cost per simulation run even further than you mentioned:

1) use multiple simulations per simulation run, with the `.step` directive.

2) use the simulator in-process. 
 Explanation: Spicelib calls the simulator as an external executable, and therefore you will have the cost of starting the ltspice executable as an independent process. 
 If the simulator were available as as shared library, we might be able lower the startup cost for each individual simulation run if you call multiple simulations from the same python process. 
 Now LTspice is not available as a shared library. However, ngspice and xyce can be exposed via shared libraries, and on top of that, there is a python interface library for them (PySpice). We have not (yet) interfaced with that library, but if it is considered interesting, we could try to interface with it. It uses a totally different approach to editing netlists though, so we'd have to decide how that integration would turn out. Or we could try to directly interface with the shared libraries. With the same caveats that PySpice mentions: (for ngspice:) `this approach is less safe than the subprocess approach. In case of bugs within Ngspice, we can expect some side effects like memory leaks or worse unexpected things.` You'll have a more close relation to the simulator, but it comes with pros and cons.

closed 2025-03-29 by DavitKharshiladzeTrain

nunobrum, 2025-03-30:
If you want to process the data immediately, the idea is that you would rather use the run_now() instead of run().
It has the behaviour you want.  One simulation fully executed at a time.
The idea of the run() and wait_completion() would be to have the process the most asynchronous as possible. This way, having the possibility of launching a second simulation before the first one is completed.
For that the wait_completion() should be outside the loop.  
For processing the data, you can either do it after the loop has ended, or, you can use a callback function that will process it.
The advantage of the callback function, is that the processing is also done in parallel.