nunobrum/spicelib #201: FailureMode code errors?

hb020, 2025-04-07:
Inside `FailureMode` I found the following discrepancies:

in `FailureMode.__init__`:
```python
SimAnalysis.__init__(self, circuit_file, simulator, runner)
```

while `SimAnalysis.__init__` signature is
```python
 def __init__(self, circuit_file: Union[str, BaseEditor], runner: Optional[AnyRunner] = None):
```
No `simulator` there.

And inside the `FailureMode` class, all but 1 run statements refer to `simulator`, not to `runner`.

In the absence of example code, test cases or sphinx documentation regarding this class, I hesitate to correct this, as this looks work in progress (but abandoned). Your feedback on how to move forward with this?

nunobrum, 2025-04-07:
FailureMode is indeed a stalled implementation. 
The idea was to be able to make an automatic component level FMEA. That is, replace at a time, each component by all its failure modes and check the impact on the circuit.
For example, a capacitor may short or become an open circuit (also depending if X or Y type). Then looking at the inputs /outputs /supplies, we can see whether the failure is contained or it it propagates.

This was something I was interested in when I was making projects for space. Meanwhile, moved on to other domains (regret 😿 ).

nunobrum, 2025-04-16:
Removed stalled failure_modes.py file

closed 2025-04-16 by nunobrum