nunobrum/spicelib #213: ngspice netlist with osdi model compiled by openvaf can not use spicelib to simulate

LeeBooa, 2025-05-27:
hello, can the spicelib module use to simulate the ngspice netlist with osdi model compiled by openvaf？i try to do above work, but the spicelib module always show me the error "unrecognized command in line："N1 C B 0 0 npn_full_sh"", the npn_full_sh model is the model compiled by openvaf.
thank you very much.🌹

hb020, 2025-05-27:
Well, openvaf and MacOS do not play well together it seems. I will need to set that up on Linux, but as far as I see, no extra changes in spicelib would be needed: osdi compatibility is a ngspice system wide setting, not controlled per simulation run.

@LeeBooa can you see if the code in the attached branch (https://github.com/nunobrum/spicelib/tree/213-support-verilog-elements) is OK for you? It should, adding the N type was pretty easy.

nunobrum, 2025-07-28:
All seems OK.

closed 2025-07-28 by nunobrum